jiangdongguo/AndroidUSBCamera #496: 拍摄的视频显示超级大，也不显示时长

EthanCo, 2022-11-17:
运行demo
华为MatePad Pro 拍摄的视频显示超级大，也不显示时长。
但是我用手机一加7Pro 拍摄的是正常的，视频大小和时长都有显示，就很奇怪。
运行日本的[saki4510t/UVCCamera](https://github.com/saki4510t/UVCCamera)华为MatePad Pro存储的视频显示的大小和时长也都是正常的。

EthanCo, 2022-11-17:
我在华为MatePad Pro中用MxPlayer查看视频，也是能查看到真实大小和时长的，华为那个相册就很不正常。

EthanCo, 2022-11-17:
我试了一下华为P40 (同样是HarmonyOS 2.0.0系统)，发现也有这个问题。
但用它内置的文件管理器，查看到的视频大小和时长又是正常的。
只有它的那个相册显示视频是超级大，不显示时长。

EthanCo, 2022-11-17:
我知道了，相机里的视频大小和时长是从`MediaStore`数据库中读取出来的
```java
val sizeIndex = getColumnIndex(MediaStore.MediaColumns.SIZE)
val durationIndex = getColumnIndex(MediaStore.Video.Media.DURATION)
```
数据库里存的有问题，那为什么存储会有问题呢 ?

EthanCo, 2022-11-17:
我在自己项目中，通过其他方式获取时长和大小 临时处理 了，但是相册里还是显示的不正常的值。
```java
val size = File(path).length()
val duration = getLocalVideoDuration(path)

//返回视频时长 (毫秒)
private fun getLocalVideoDuration(filePath: String?): Long {
    var duration = 0L
    duration = try {
        val mmr = MediaMetadataRetriever()
        mmr.setDataSource(filePath)
        mmr.extractMetadata(MediaMetadataRetriever.METADATA_KEY_DURATION).toLong()
    } catch (e: Exception) {
        e.printStackTrace()
        return duration
    }
    return duration
}
```

EthanCo, 2022-11-17:
我知道原因了
`Mp4Muxer.kt`中，向`MediaStore`数据库插入数据的时候，`getVideoContentValues()`代码有错误的地方  
```java
values.put(MediaStore.Video.Media.SIZE, file.totalSpace)
```
实际上应该是
```java
values.put(MediaStore.Video.Media.SIZE, file.length())
```
还有，缺少`duration`这一项，所以`duration`读取到的是`0`，导致相册视频时长无法显示  
还需要加上这个
```
values.put(MediaStore.Video.Media.DURATION, getLocalVideoDuration(file.path))

//返回视频时长 (毫秒)
private fun getLocalVideoDuration(filePath: String?): Long {
    return try {
        val mmr = MediaMetadataRetriever()
        mmr.setDataSource(filePath)
        mmr.extractMetadata(MediaMetadataRetriever.METADATA_KEY_DURATION)?.toLong() ?:0L
    } catch (e: Exception) {
        e.printStackTrace()
        0L
    }
}
```

EthanCo, 2022-11-17:
由于[saki4510t/UVCCamera](https://github.com/saki4510t/UVCCamera)没有到`MediaStore`，所以不存在这个问题。

jiangdongguo, 2022-11-21:
好的，迟点我验证下，OK的话就更新上去。谢谢支持！

closed 2023-02-14 by jiangdongguo