jiangdongguo/AndroidUSBCamera #501: unsupported preview size

OnesangYou, 2022-11-30:
uvc camera does not work in 3.2.10 version. (It works with 3.2.8 version)

022-11-30 21:30:50.964 12467-12545/com.jiangdg.ausbc I/libusb/core: [12545*core.c:2085:libusb_init2]:libusb v1.0.19.10903
2022-11-30 21:30:50.965 12467-12545/com.jiangdg.ausbc I/libusb/usbfs: [12545*android_usbfs.c:504:op_init2]:start up hotplug event handler
2022-11-30 21:30:50.965 12467-12545/com.jiangdg.ausbc I/libusb/usbfs: [12545*android_usbfs.c:512:op_init2]:call android_scan_devices
2022-11-30 21:30:50.965 12467-12545/com.jiangdg.ausbc I/AUSBC: [CameraUvc]  getAllPreviewSizes = []
2022-11-30 21:30:50.966 12467-12545/com.jiangdg.ausbc E/AUSBC: [CameraUvc]   unsupported preview size(640, 480)
2022-11-30 21:30:50.966 12467-12467/com.jiangdg.ausbc I/AUSBC: [CameraUvc]  getAllPreviewSizes = []
2022-11-30 21:30:50.966 12467-12552/com.jiangdg.ausbc E/libUVCCamera: [12552*UVCPreview.cpp:524:prepare_preview]:could not negotiate with camera:err=-51

chensijie0725, 2023-02-14:
尝试一下修改获取数据的格式，把yuv改成MJPEG，好像默认就是去yuv的，但是会取不到支持的分辨率

jiangdongguo, 2023-02-15:
update to new on branch master and have a try

imamabdulazis, 2024-03-24:
Any update regarding this? I was using 3.3.3 version but the screen is still blank.

```
 [10151*UVCCamera.cpp:172:connect]:could not open camera:err=-99
2024-03-24 22:30:33.454 32453-10152 CameraUvc               pid-32453                            I  getAllPreviewSizes = []
2024-03-24 22:30:33.456 32453-10151 CameraUvc               pid-32453                            E   preview failed, err = open failed:result=-99----->id_camera=516438493696;venderId=3804;productId=8272;fileDescriptor=186;busNum=1;devAddr=2;usbfs=/dev/bus/usb
                                                                                                    Exception：调用nativeConnect返回值：-99
                                                                                                    java.lang.UnsupportedOperationException: open failed:result=-99----->id_camera=516438493696;venderId=3804;productId=8272;fileDescriptor=186;busNum=1;devAddr=2;usbfs=/dev/bus/usb
                                                                                                    Exception：调用nativeConnect返回值：-99
                                                                                                    	at com.jiangdg.uvc.UVCCamera.open(UVCCamera.java:217)
                                                                                                    	at com.jiangdg.ausbc.camera.CameraUvcStrategy.createCamera(CameraUvcStrategy.kt:133)
                                                                                                    	at com.jiangdg.ausbc.camera.CameraUvcStrategy.startPreviewInternal(CameraUvcStrategy.kt:114)
                                                                                                    	at com.jiangdg.ausbc.camera.ICameraStrategy.handleMessage(ICameraStrategy.kt:103)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                                    	at android.os.Looper.loop(Looper.java:246)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2024-03-24 22:30:33.457 32453-10152 CameraUvc               pid-32453                            E   unsupported preview size(1280, 720)
--------- beginning of crash
2024-03-24 22:30:33.461 32453-10154 libUVCCamera            pid-32453                            E  [10154*UVCPreview.cpp:524:prepare_preview]:could not negotiate with camera:err=-51
```