jiangdongguo/AndroidUSBCamera #506: FrameProcessor

openmotion, 2022-12-09:
hello

how to add a callback to get each frame 

i have another library otaliastudio

and i use this function
`
  mCameraView.addFrameProcessor(new FrameProcessor() {
                @Override
                @WorkerThread
                public void process(@NonNull Frame frame) {`

thanks

jiangdongguo, 2023-02-13:
setPreviewCallback

Foufure13, 2023-06-09:
> setPreviewCallback

can you give the full example ?
in DemoFragment.kt
getCameraClient() function

i can add mox.addPreviewDataCallBack()
but i don't have an example for : callBack: IPreviewDataCallBack

do you have it ?
thanks

openmotion, 2023-06-14:
i created this function 
StartPreviewCallBack

```
 private fun StartPreviewCallBack() {
        Log.d(TAG, "StartPreviewCallBack start")
        addPreviewDataCallBack( object :  IPreviewDataCallBack  {
            override fun onPreviewData(data: ByteArray?, format: IPreviewDataCallBack.DataFormat) {
                TODO("Not yet implemented")
                Log.d(TAG, "onPreviewData ??")
            }
        })
    }
```
but no preview data in my log
i added this in DemoFragment.kt and extended by : ,IPreviewDataCallBack