jiangdongguo/AndroidUSBCamera #510: DataBinding breaks Library

JMonson-SoftPoint, 2023-01-07:
When attempting to use this library along with a DataBinding one of the objects that is expected to be a LifecycleOwner is not causing the library to crash. If the owning activity is not a binding but the fragment is then it works as expected

`FATAL EXCEPTION: main
                                                                                                    Process: us.softpoint.kioskpoint, PID: 22160
                                                                                                    java.lang.IllegalArgumentException: context should be subclass of LifecycleOwner!
                                                                                                    	at com.jiangdg.ausbc.CameraClient.<init>(CameraClient.kt:84)
                                                                                                    	at com.jiangdg.ausbc.CameraClient$Builder.build(CameraClient.kt:745)
                                                                                                    	at com.jiangdg.ausbc.base.CameraFragment.getDefault(CameraFragment.kt:415)
                                                                                                    	at com.jiangdg.ausbc.base.CameraFragment.initData(CameraFragment.kt:47)
                                                                                                    	at com.jiangdg.ausbc.base.BaseFragment.onViewCreated(BaseFragment.kt:45)
                                                                                                    	at androidx.fragment.app.Fragment.performViewCreated(Fragment.java:2987)
                                                                                                    	at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:546)
                                                                                                    	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:282)
                                                                                                    	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:112)
                                                                                                    	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1647)
                                                                                                    	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:3128)
                                                                                                    	at androidx.fragment.app.FragmentManager.dispatchActivityCreated(FragmentManager.java:3072)
                                                                                                    	at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:251)
                                                                                                    	at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:502)
                                                                                                    	at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:248)
                                                                                                    	at com.softpointdev.commonpoint.common.ui.BaseActivity.onStart(BaseActivity.java:56)
                                                                                                    	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1510)
                                                                                                    	at android.app.Activity.performStart(Activity.java:8590)
                                                                                                    	at android.app.ActivityThread.handleStartActivity(ActivityThread.java:4196)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:221)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:201)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:173)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2576)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8775)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)`

JMonson-SoftPoint, 2023-01-07:
After a lot of experimenting, I have managed to figure out that Hilt when using an AndroidEntryPoint annotation on the Fragment will cause the library to break. Not quite sure why that is the case but that is what is happening

jiangdongguo, 2023-02-11:
Hi, please update to 3.3.+(new is 3.3.2) and have try.