jiangdongguo/AndroidUSBCamera #511: 多个USB摄像头，如何设置初始显示的摄像头呢 ?

EthanCo, 2023-01-10:
设置`CameraRequest.Builder().setCameraId()`似乎没有用

MarkThunder, 2023-01-17:
> 设置`CameraRequest.Builder().setCameraId()`似乎没有用

我也设置了，也没有效果，请问您解决了吗？

Petri-Leinonen, 2023-01-27:
I'm also having trouble setting a specific camera. The `setCameraId()` doesn't do anything.

MarkThunder, 2023-01-28:
> I'm also having trouble setting a specific camera. The `setCameraId()` doesn't do anything.

I solved it by modifying CameraUvcStrategy.kt

Petri-Leinonen, 2023-01-28:
> > I'm also having trouble setting a specific camera. The `setCameraId()` doesn't do anything.
> 
> I solved it by modifying CameraUvcStrategy.kt

Can you please show what modifications you made to make it work?

MarkThunder, 2023-01-28:
> > > I'm also having trouble setting a specific camera. The `setCameraId()` doesn't do anything.
> > 
> > 
> > I solved it by modifying CameraUvcStrategy.kt
> 
> Can you please show what modifications you made to make it work?

Sorry, it's time for a break in China, I'll reply tomorrow

Modify section
```java
            /**
             * Called by granted permission
             *
             * @param device usb device info,see [UsbDevice]
             */
            override fun onConnect(
                device: UsbDevice?,
                ctrlBlock: UsbControlBlock?,
                createNew: Boolean
            ) {
                if (Utils.debugCamera) {
                    Logger.i(TAG, "onConnect device = ${device?.deviceName}")
                }
                if (!isUsbCamera(device) && !isFilterDevice(
                        getContext(),
                        device
                    ) && !mCacheDeviceList.contains(device)
                ) {
                    return
                }
                mDevSettableFuture = SettableFuture()
                mCtrlBlockSettableFuture = SettableFuture()
                getRequest()?.apply {
                    if (getSurfaceTexture() != null) {
                        startPreview(this, getSurfaceTexture())
                    } else {
                        startPreview(this, getSurfaceHolder())
                    }
                }
                if (pid == 0 || vid == 0) {
                    Log.d(TAG, "onConnect: pid$pid,vid$vid")
                    mDevSettableFuture?.set(device)
                    mCtrlBlockSettableFuture?.set(ctrlBlock)
                } else {
                    // todo
                    // 可以增加配置入口，修改pid, vid
                    mCacheDeviceList.forEach { dev ->
                        if (dev.vendorId == vid && dev.productId == pid) {
                            mDevSettableFuture?.set(dev)
                            mCtrlBlockSettableFuture?.set(UsbControlBlock(mUsbMonitor, dev))
                        }
                    }
                }
                mConnectSettableFuture.set(true)
            }
```

jiangdongguo, 2023-02-11:
Hi, please update to 3.3.+(new is 3.3.2) ， read ReadMe.txt in detail and have try.

closed 2023-02-14 by jiangdongguo