jiangdongguo/AndroidUSBCamera #512: 分辨率高于1280*720，帧率变低，预览画面卡顿

DaveZF, 2023-01-10:
问题描述：demo工程APK或自定义开发APK，在分辨率高于1280*720后，掉帧严重画面卡顿，帧率仅有10fps左右；低帧率往下如1024x768 、800x600均能达到60fps，

版本：3.2.10 （demo 工程APK验证）
运行环境：华为Mate40E pro (harmonyOS 3.0 )  荣耀V20
USB设备：绿联采集卡（HDMI2USB）
设备支持分辨率：
`getAllPreviewSizes = [Size(1920x1080@ 0.0,type:4,frame:0,index:0,[]), Size(1600x1200@ 0.0,type:4,frame:0,index:1,[]), Size(1360x768@ 0.0,type:4,frame:0,index:2,[]), Size(1280x1024@ 0.0,type:4,frame:0,index:3,[]), Size(1280x960@ 0.0,type:4,frame:0,index:4,[]), Size(1280x720@ 0.0,type:4,frame:0,index:5,[]), Size(1024x768@ 0.0,type:4,frame:0,index:6,[]), Size(800x600@ 0.0,type:4,frame:0,index:7,[]), Size(720x576@ 0.0,type:4,frame:0,index:8,[]), Size(720x480@ 0.0,type:4,frame:0,index:9,[]), Size(640x480@ 0.0,type:4,frame:0,index:10,[])]`

截取部分LOG：
**1280x720**
`
2023-01-10 11:14:55.807 I/AUSBC: [CameraUvc]   createCamera success! request = CameraRequest(previewWidth=1280, previewHeight=720, cameraId='2002', isFrontCamera=false, isContinuousAFModel=true, isContinuousAEModel=true)
2023-01-10 11:14:55.808 I/AUSBC: [CameraUvc]   start preview success!!!, id(/dev/bus/usb/002/002

2023-01-10 11:14:55.959 I/AUSBC: [RenderManager]  camera render frame rate is 6 fps
2023-01-10 11:14:56.965 I/AUSBC: [RenderManager]  camera render frame rate is 61 fps
2023-01-10 11:14:57.965 I/AUSBC: [RenderManager]  camera render frame rate is 60 fps
2023-01-10 11:14:58.965 I/AUSBC: [RenderManager]  camera render frame rate is 60 fps
2023-01-10 11:14:59.966 I/AUSBC: [RenderManager]  camera render frame rate is 60 fps
2023-01-10 11:15:00.971 I/AUSBC: [RenderManager]  camera render frame rate is 60 fps
2023-01-10 11:15:01.983 I/AUSBC: [RenderManager]  camera render frame rate is 60 fps
2023-01-10 11:15:02.993 I/AUSBC: [RenderManager]  camera render frame rate is 60 fps
2023-01-10 11:15:04.090 I/AUSBC: [RenderManager]  camera render frame rate is 53 fps`

**1280x960**
`2023-01-10 11:13:55.429 I/AUSBC: [CameraClient]  updateResolution size = 1280x960

2023-01-10 11:13:55.429 I/AUSBC: [CameraClient]  closeCamera...
2023-01-10 11:13:55.436 I/AUSBC: [CameraUvc]  stopPreviewInternal
2023-01-10 11:13:55.438 I/AUSBC: [AbstractRender]  release surface texture render success!
2023-01-10 11:13:55.440 I/AUSBC: [EGLEvn]  Release EGL Success!
2023-01-10 11:13:55.441 I/AUSBC: [AbstractRender]  release surface texture render success!
2023-01-10 11:13:55.441 I/AUSBC: [AbstractRender]  release surface texture render success!
2023-01-10 11:13:55.506 I/AUSBC: [USBMonitor]  UsbControlBlock#close:
2023-01-10 11:13:55.507 I/AUSBC: [CameraUvc]  onDisconnect device = /dev/bus/usb/002/002
2023-01-10 11:13:55.507 V/AUSBC: [UVCCamera]  close:finished
2023-01-10 11:13:55.930 I/AUSBC: [CameraClient]  Audio record by using system mic
2023-01-10 11:13:55.930 I/AUSBC: [CameraClient]  start open camera request = CameraRequest(previewWidth=1280, previewHeight=960, cameraId='2002', isFrontCamera=false, isContinuousAFModel=true, isContinuousAEModel=true), gl = true
2023-01-10 11:13:55.931 I/AUSBC: [AspectRatioTextureView]  AspectRatio = 1.3333333333333333
2023-01-10 11:13:55.946 I/AUSBC: [AbstractRender]  create external texture, id = 0
2023-01-10 11:13:55.950 I/AUSBC: [CameraUvc]   find some device list, = [/dev/bus/usb/002/002]
2023-01-10 11:13:55.952 I/AUSBC: [EGLEvn]  Init EGL Success!
2023-01-10 11:13:55.952 I/AUSBC: [EGLEvn]  setupSurface Success!
2023-01-10 11:13:55.954 I/AUSBC: [AbstractRender]  init surface texture render success!
2023-01-10 11:13:55.956 I/AUSBC: [AbstractRender]  init surface texture render success!
2023-01-10 11:13:55.957 I/AUSBC: [AbstractRender]  init surface texture render success!
2023-01-10 11:13:55.958 I/AUSBC: [AbstractRender]  create texture, id = 1
2023-01-10 11:13:55.958 I/AUSBC: [AbstractRender]  create texture, id = 2
2023-01-10 11:13:55.960 V/AUSBC: [UVCCamera]  close:finished
2023-01-10 11:13:56.017 I/AUSBC: [CameraUvc]  getAllPreviewSizes = [Size(1920x1080@ 0.0,type:4,frame:0,index:0,[]), Size(1600x1200@ 0.0,type:4,frame:0,index:1,[]), Size(1360x768@ 0.0,type:4,frame:0,index:2,[]), Size(1280x1024@ 0.0,type:4,frame:0,index:3,[]), Size(1280x960@ 0.0,type:4,frame:0,index:4,[]), Size(1280x720@ 0.0,type:4,frame:0,index:5,[]), Size(1024x768@ 0.0,type:4,frame:0,index:6,[]), Size(800x600@ 0.0,type:4,frame:0,index:7,[]), Size(720x576@ 0.0,type:4,frame:0,index:8,[]), Size(720x480@ 0.0,type:4,frame:0,index:9,[]), Size(640x480@ 0.0,type:4,frame:0,index:10,[])]
2023-01-10 11:13:56.051 I/AUSBC: [CameraUvc]   createCamera success! request = CameraRequest(previewWidth=1280, previewHeight=960, cameraId='2002', isFrontCamera=false, isContinuousAFModel=true, isContinuousAEModel=true)
2023-01-10 11:13:56.052 I/AUSBC: [CameraUvc]   start preview success!!!, id(/dev/bus/usb/002/002
2023-01-10 11:13:58.114 I/AUSBC: [RenderManager]  camera render frame rate is 21 fps
2023-01-10 11:13:59.421 I/AUSBC: [RenderManager]  camera render frame rate is 5 fps
2023-01-10 11:14:01.047 I/AUSBC: [RenderManager]  camera render frame rate is 3 fps
2023-01-10 11:14:03.985 I/AUSBC: [RenderManager]  camera render frame rate is 6 fps`

**1920x1080**
`2023-01-10 11:14:32.989 I/AUSBC: [CameraClient]  updateResolution size = 1920x1080

2023-01-10 11:14:32.989 I/AUSBC: [CameraClient]  closeCamera...
2023-01-10 11:14:32.994 I/AUSBC: [AbstractRender]  release surface texture render success!
2023-01-10 11:14:32.996 I/AUSBC: [CameraUvc]  stopPreviewInternal
2023-01-10 11:14:32.996 I/AUSBC: [EGLEvn]  Release EGL Success!
2023-01-10 11:14:32.996 I/AUSBC: [AbstractRender]  release surface texture render success!
2023-01-10 11:14:32.996 I/AUSBC: [AbstractRender]  release surface texture render success!
2023-01-10 11:14:33.075 I/AUSBC: [USBMonitor]  UsbControlBlock#close:
2023-01-10 11:14:33.076 I/AUSBC: [CameraUvc]  onDisconnect device = /dev/bus/usb/002/002
2023-01-10 11:14:33.076 V/AUSBC: [UVCCamera]  close:finished
2023-01-10 11:14:33.491 I/AUSBC: [CameraClient]  Audio record by using system mic
2023-01-10 11:14:33.493 I/AUSBC: [CameraClient]  start open camera request = CameraRequest(previewWidth=1920, previewHeight=1080, cameraId='2002', isFrontCamera=false, isContinuousAFModel=true, isContinuousAEModel=true), gl = true
2023-01-10 11:14:33.494 I/AUSBC: [AspectRatioTextureView]  AspectRatio = 1.7777777777777777
2023-01-10 11:14:33.505 I/AUSBC: [AbstractRender]  create external texture, id = 0
2023-01-10 11:14:33.507 I/AUSBC: [CameraUvc]   find some device list, = [/dev/bus/usb/002/002]
2023-01-10 11:14:33.509 I/AUSBC: [EGLEvn]  Init EGL Success!
2023-01-10 11:14:33.509 I/AUSBC: [EGLEvn]  setupSurface Success!
2023-01-10 11:14:33.512 I/AUSBC: [AbstractRender]  init surface texture render success!
2023-01-10 11:14:33.514 I/AUSBC: [AbstractRender]  init surface texture render success!
2023-01-10 11:14:33.516 V/AUSBC: [UVCCamera]  close:finished
2023-01-10 11:14:33.517 I/AUSBC: [AbstractRender]  init surface texture render success!
2023-01-10 11:14:33.518 I/AUSBC: [AbstractRender]  create texture, id = 1
2023-01-10 11:14:33.518 I/AUSBC: [AbstractRender]  create texture, id = 2
2023-01-10 11:14:33.578 I/AUSBC: [CameraUvc]  getAllPreviewSizes = [Size(1920x1080@ 0.0,type:4,frame:0,index:0,[]), Size(1600x1200@ 0.0,type:4,frame:0,index:1,[]), Size(1360x768@ 0.0,type:4,frame:0,index:2,[]), Size(1280x1024@ 0.0,type:4,frame:0,index:3,[]), Size(1280x960@ 0.0,type:4,frame:0,index:4,[]), Size(1280x720@ 0.0,type:4,frame:0,index:5,[]), Size(1024x768@ 0.0,type:4,frame:0,index:6,[]), Size(800x600@ 0.0,type:4,frame:0,index:7,[]), Size(720x576@ 0.0,type:4,frame:0,index:8,[]), Size(720x480@ 0.0,type:4,frame:0,index:9,[]), Size(640x480@ 0.0,type:4,frame:0,index:10,[])]
2023-01-10 11:14:33.612 I/AUSBC: [CameraUvc]   createCamera success! request = CameraRequest(previewWidth=1920, previewHeight=1080, cameraId='2002', isFrontCamera=false, isContinuousAFModel=true, isContinuousAEModel=true)
2023-01-10 11:14:33.613 I/AUSBC: [CameraUvc]   start preview success!!!, id(/dev/bus/usb/002/002
2023-01-10 11:14:38.827 I/AUSBC: [RenderManager]  camera render frame rate is 2 fps
2023-01-10 11:14:39.856 I/AUSBC: [RenderManager]  camera render frame rate is 11 fps`

jiangdongguo, 2023-02-11:
试试3.3.2 开启opengl呢

DaveZF, 2023-02-24:
> 

> 试试3.3.2 开启opengl呢

在最新版本3.3.3上使用Demo，也是一样的问题，帧率在10几帧。
猜测应该是获取原始数据出现卡顿，

2023-01-30 12:01:52.299 19455-27543 libusb/core             com.jiangdg.ausbc                    I  [27543*core.c:2085:libusb_init2]:libusb v1.0.19.10903
2023-01-30 12:01:52.299 19455-27543 libusb/usbfs            com.jiangdg.ausbc                    I  [27543*android_usbfs.c:504:op_init2]:start up hotplug event handler
2023-01-30 12:01:52.299 19455-27543 libusb/usbfs            com.jiangdg.ausbc                    I  [27543*android_usbfs.c:512:op_init2]:call android_scan_devices
2023-01-30 12:01:52.304 19455-27543 libuvc/device           com.jiangdg.ausbc                    E  [27543*device.c:330:uvc_open]:internal_devh->info->ctrl_if.bEndpointAddress is null
2023-01-30 12:01:52.317 19455-27543 AUSBC                   com.jiangdg.ausbc                    I  [CameraUVC]  aspect ratio = null, getAllPreviewSizes = [PreviewSize(width=1920, height=1080), PreviewSize(width=1600, height=1200), PreviewSize(width=1360, height=768), PreviewSize(width=1280, height=1024), PreviewSize(width=1280, height=960), PreviewSize(width=1280, height=720), PreviewSize(width=1024, height=768), PreviewSize(width=800, height=600), PreviewSize(width=720, height=576), PreviewSize(width=720, height=480), PreviewSize(width=640, height=480)], 
2023-01-30 12:01:52.317 19455-27543 AUSBC                   com.jiangdg.ausbc                    I  [CameraUVC]  getSuitableSize: PreviewSize(width=1280, height=960)
2023-01-30 12:01:52.319 19455-27543 AUSBC                   com.jiangdg.ausbc                    I  [CameraUVC]  aspect ratio = null, getAllPreviewSizes = [PreviewSize(width=1920, height=1080), PreviewSize(width=1600, height=1200), PreviewSize(width=1360, height=768), PreviewSize(width=1280, height=1024), PreviewSize(width=1280, height=960), PreviewSize(width=1280, height=720), PreviewSize(width=1024, height=768), PreviewSize(width=800, height=600), PreviewSize(width=720, height=576), PreviewSize(width=720, height=480), PreviewSize(width=640, height=480)], 
2023-01-30 12:01:52.319 19455-27543 AUSBC                   com.jiangdg.ausbc                    I  [MultiCameraClient]  Audio record by using device internal mic
2023-01-30 12:01:52.324 19455-27543 AUSBC                   com.jiangdg.ausbc                    I  [CameraUVC]   start preview, name = /dev/bus/usb/002/008, preview=PreviewSize(width=1280, height=960)
2023-01-30 12:01:52.324 19455-27543 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  create camera SurfaceTexture: android.graphics.SurfaceTexture@b17077e
2023-01-30 12:01:52.325 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [AbstractRender]  create texture, id = 2
2023-01-30 12:01:52.325 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [AbstractFboRender]  load fbo, textures: [I@68ac4df, buffers: [I@48bf52c
2023-01-30 12:01:52.325 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [AbstractRender]  create texture, id = 3
2023-01-30 12:01:52.325 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [AbstractFboRender]  load fbo, textures: [I@52fbf5, buffers: [I@981d8a
2023-01-30 12:01:52.330 19455-27570 libUVCCamera            com.jiangdg.ausbc                    I  [27570*UVCPreview.cpp:509:prepare_preview]:frameSize=(1280,960)@MJPEG
2023-01-30 12:01:52.455 19455-27573 libUVCCamera            com.jiangdg.ausbc                    W  [27573*UVCPreview.cpp:128:get_frame]:allocate new frame
2023-01-30 12:01:52.455 19455-27570 libUVCCamera            com.jiangdg.ausbc                    W  [27570*UVCPreview.cpp:128:get_frame]:allocate new frame
2023-01-30 12:01:53.893 19455-27573 libUVCCamera            com.jiangdg.ausbc                    W  [27573*UVCPreview.cpp:128:get_frame]:allocate new frame
2023-01-30 12:01:53.910 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 1 fps-->gl_render
2023-01-30 12:01:54.023 19455-27573 libUVCCamera            com.jiangdg.ausbc                    W  [27573*UVCPreview.cpp:128:get_frame]:allocate new frame
2023-01-30 12:01:57.143 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 6 fps-->gl_render
2023-01-30 12:02:00.400 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 6 fps-->gl_render
2023-01-30 12:02:03.634 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 6 fps-->gl_render
2023-01-30 12:02:06.863 19455-27550 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 6 fps-->gl_render

ronpetit, 2024-04-11:
@jiangdongguo Same issue here, 1080p with opengl will always be on 2fps - 10fps, and skipping a lot of frames, happens on my app and on the demo APP with the 3.3.3 version

heacsing, 2024-05-20:
same problem. And after decreasing `MAX_FPS`, the dropping rate of render frame would be a bit slow down.

JeffZane, 2024-11-01:
请问有人解决吗？华为Mate60和P60都遇到了，首次进入预览640x480正常，切换其他分辨率直接黑屏，重启App也不行，在demo apk上一样的现象

hiboxgit, 2024-11-18:
本来我也遇到了这个问题，分辨率提高，帧率直线下降，并且延迟很高，一直寻求解决方案。。。结果求人不如求己啊。。。
我知道怎么搞了。。。这里面涉及两个问题，一个是延迟高，一个是帧率低，我之前一直搞混淆了，一致关注预览卡顿问题，跑偏了。

muselee, 2024-11-22:
> 本来我也遇到了这个问题，分辨率提高，帧率直线下降，并且延迟很高，一直寻求解决方案。。。结果求人不如求己啊。。。 我知道怎么搞了。。。这里面涉及两个问题，一个是延迟高，一个是帧率低，我之前一直搞混淆了，一致关注预览卡顿问题，跑偏了。

您好，怎么搞定了，能指导下吗

hiboxgit, 2024-12-25:
怎么指导。。

liyx2018, 2025-01-20:
> 怎么指导。。

大佬 怎么搞定的  需要修改哪些代码