jiangdongguo/AndroidUSBCamera #515: Change Preview size

JulesD29, 2023-01-17:
Hi,
I would like to customise the preview dispalyed size.

In my ***camera_view.xml*** I writed the following lines :

```
  <FrameLayout
        android:id="@+id/cameraViewContainer"
        
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent"

        android:layout_width="wrap_content"
        android:layout_height="wrap_content"/>
```

I also created the ***Camera.kt*** with these lines :

```
    override fun getCameraClient(): CameraClient {
        mCameraClient = CameraClient.newBuilder(requireContext())
            .setEnableGLES(true)   // use opengl render
            .setRawImage(true)     // capture raw or filter image
            //.setDefaultEffect(EffectBlackWhite(requireContext())) // default effect
            .setCameraStrategy(CameraUvcStrategy(requireContext())) // camera type
            .setCameraRequest(getCameraRequest()) // camera configurations
            .setDefaultRotateType(RotateType.ANGLE_270) // default camera rotate angle
            .openDebug(true) // is debug mode
            .build()
        return mCameraClient
    }

    private fun getCameraRequest(): CameraRequest {
        return CameraRequest.Builder()
            .setFrontCamera(false) // only for camera1/camera2
            .setPreviewWidth(1280)
            .setPreviewHeight(720)
            .create()
    }
```

I would like to have the screen with all the preview displayed, but by doing as above, the screen is displayed only on a small part of the screen.
How do I change the preview resolution? Because the .setPreview..... method does not work.

For info, I get this in the logs about CameraClient:

```
---------------------------- PROCESS STARTED (6244) for package com.example.quick_air ----------------------------
2023-01-17 15:24:13.167  6244-6244  CameraClient            com.example.quick_air                I  init camera client, camera = com.jiangdg.ausbc.camera.CameraUvcStrategy@733b7c0
2023-01-17 15:24:13.173  6244-6244  CameraClient            com.example.quick_air                I  init camera client, camera = com.jiangdg.ausbc.camera.CameraUvcStrategy@63f4231
2023-01-17 15:24:14.183  6244-6244  CameraClient            com.example.quick_air                I  Audio record by using system mic
2023-01-17 15:24:14.185  6244-6244  CameraClient            com.example.quick_air                I  start open camera request = CameraRequest(previewWidth=1280, previewHeight=720, cameraId='', isFrontCamera=false, isContinuousAFModel=true, isContinuousAEModel=true), gl = true
2023-01-17 15:24:14.223  6244-6244  CameraClient            com.example.quick_air                I  Display render, width=377, height=671
2023-01-17 15:24:22.941  6244-6244  CameraClient            com.example.quick_air                I  closeCamera...
2023-01-17 15:24:22.944  6244-6244  CameraClient            com.example.quick_air                I  closeCamera...
2023-01-17 15:24:22.947  6244-6244  CameraClient            com.example.quick_air                I  closeCamera...
```

Have you ever faced this problem?
thanks
Jules

casper4738, 2023-02-08:
"the screen is displayed only on a small part of the screen " <-- me too 
 after use v3.2.8, its normal

JulesD29, 2023-02-08:
Yeah but I use the version 3.2.10 :
`    implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.2.10'
`
How we can fix that size preview ? Go on a previous version ?

1369521908, 2023-02-09:
the CameraRequest size is the camera size
but on your ui is the other size
[image]

jiangdongguo, 2023-02-11:
Hi, please update to 3.3.+(new is 3.3.2) and have try.