jiangdongguo/AndroidUSBCamera #520: Frame is zero

idan123456, 2023-02-07:
Hey, 
For some reason in most of the time I connect the camera while running the app it doesn't sho an image,
I've noticed that the frame for all resolution is 0.
I will note that in some cases that I connected the camera it worked fine with high fps.

> I/CameraUvc: getAllPreviewSizes = [Size(3264x2448@ 0.0,type:4,frame:0,index:0,[]), Size(1920x1080@ 0.0,type:4,frame:0,index:1,[]), Size(1600x1200@ 0.0,type:4,frame:0,index:2,[]), Size(1280x720@ 0.0,type:4,frame:0,index:3,[]), Size(800x600@ 0.0,type:4,frame:0,index:4,[]), Size(640x480@ 0.0,type:4,frame:0,index:5,[])]
I/libUVCCamera: [31911*UVCPreview.cpp:285:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV20SP:

Did someone else encountered this? how can I fix it?
is it possible to set the frame manually?

jiangdongguo, 2023-02-11:
Hi, please update to 3.3.+(new is 3.3.2) and have try.

idan123456, 2023-02-12:
> Hi, please update to 3.3.+(new is 3.3.2) and have try.

It fixed the problem thank you.
Also, I need to set the FPS to 10 is there a way to set it manually?

jiangdongguo, 2023-02-13:
unsupported now