jiangdongguo/AndroidUSBCamera #521: Multiple camera problem when unplugging a device

Petri-Leinonen, 2023-02-11:
I have multiple cameras  plugged on to Android tablet, through a USB hub. If I start the camera capture and unplug the active camera device from the USB hug, the capture automatically seems to transfer to another camera that is still hooked up to the USB hub.

Is it possible to stop the camera capture, if the active camera device is unplugged?

jiangdongguo, 2023-02-12:
you need use CameraFragment or CameraActivity instead of MultiCameraFragment

Petri-Leinonen, 2023-02-12:
> 

I need to support for multiple cameras, so I can switch to another one on demand. But I would like it to work so, that if I unplug the active camera, it would stop the capture, rather than automatically switch to another camera that is plugged to USB hub.

Is it possible to somehow stop capture if the active camera is unplugged from the USB hub?

jiangdongguo, 2023-02-13:
try 3.3.2