jiangdongguo/AndroidUSBCamera #522: Unsupported preview size

steagne, 2023-02-12:
Hi there!
Since i tried updating to 3.2.10 (and also to 3.3.2) it gives me the error "unsupported preview size". Basically getAllPreviewSizes gives an empty array. I also tried the 3.3.2 release app and it gives me the same error.
When using 3.2.8 the camera works successfully (getAllPreviewSizes works fine). What can i do?
Many thanks in advance!

jiangdongguo, 2023-02-13:
Please paste your 3.3.2 log txt

steagne, 2023-02-13:
`Connected to the target VM, address: 'localhost:63004', transport: 'socket'
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/libc: SetHeapTaggingLevel: tag level set to 0
I/m.jiangdg.ausb: Late-enabling -Xcheck:jni
D/ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
E/MQSEventManagerDelegate: failed to get MQSService.
W/MQSEventManagerDelegate: registerApplicationScoutThread getMQSService null
W/ActivityThread: registerApplicationScoutThread result:false
D/CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10255; state: DISABLED
W/ActivityThread: Application com.jiangdg.ausbc is waiting for the debugger on port 8100...
I/System.out: Sending WAIT chunk
D/AppScoutStateMachine: 11993-ScoutStateMachinecreated
I/System.out: Debugger has connected
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
W/MIUIScout App: Enter APP_SCOUT_WARNING State
E/MQSEventManagerDelegate: failed to get MQSService.
W/MQSEventManagerDelegate: registerApplicationScoutThread getMQSService null
W/MIUIScout App: Event:APP_SCOUT_WARNING Thread:main backtrace:
        at java.lang.Thread.sleep(Native Method)
        at java.lang.Thread.sleep(Thread.java:450)
        at java.lang.Thread.sleep(Thread.java:355)
        at android.os.Debug.waitForDebugger(Debug.java:1030)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6811)
        at android.app.ActivityThread.access$1600(ActivityThread.java:265)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2103)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loopOnce(Looper.java:210)
        at android.os.Looper.loop(Looper.java:299)
        at android.app.ActivityThread.main(ActivityThread.java:8168)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:556)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1037)
W/ActivityThread: registerApplicationScoutThread result:false
E/MQSEventManagerDelegate: failed to get MQSService.
W/MQSEventManagerDelegate: reportAppScoutEvent getMQSService null
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: debugger has settled (1369)
D/ApplicationLoaders: Returning zygote-cached class loader: /system/framework/android.test.base.jar
D/nativeloader: Configuring classloader-namespace for other apk /system/framework/org.apache.http.legacy.jar. target_sdk_version=27, uses_libraries=ALL, library_path=/data/app/~~KJTLarEP7Zv2wve1jclc7g==/com.jiangdg.ausbc-giCf1e-LM589GOxMQouxjA==/lib/arm64:/data/app/~~KJTLarEP7Zv2wve1jclc7g==/com.jiangdg.ausbc-giCf1e-LM589GOxMQouxjA==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/com.jiangdg.ausbc
D/nativeloader: Extending system_exposed_libraries: libvraudio_client.qti.so:libbinauralrenderer_wrapper.qti.so:libhoaeffects.qti.so:libQOC.qti.so:libcamera_algoup_jni.xiaomi.so:libcamera_mianode_jni.xiaomi.so:libupdateprof.qti.so:libthermalclient.qti.so:libQOC.qti.so:libdiag_system.qti.so:libqape.qti.so:libqesdk_ndk_platform.qti.so:liblistenjni.qti.so
W/ziparchive: Unable to open '/data/app/~~KJTLarEP7Zv2wve1jclc7g==/com.jiangdg.ausbc-giCf1e-LM589GOxMQouxjA==/base.dm': No such file or directory
W/ziparchive: Unable to open '/data/app/~~KJTLarEP7Zv2wve1jclc7g==/com.jiangdg.ausbc-giCf1e-LM589GOxMQouxjA==/base.dm': No such file or directory
D/nativeloader: Configuring classloader-namespace for other apk /data/app/~~KJTLarEP7Zv2wve1jclc7g==/com.jiangdg.ausbc-giCf1e-LM589GOxMQouxjA==/base.apk. target_sdk_version=27, uses_libraries=, library_path=/data/app/~~KJTLarEP7Zv2wve1jclc7g==/com.jiangdg.ausbc-giCf1e-LM589GOxMQouxjA==/lib/arm64:/data/app/~~KJTLarEP7Zv2wve1jclc7g==/com.jiangdg.ausbc-giCf1e-LM589GOxMQouxjA==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/com.jiangdg.ausbc
D/nativeloader: Extending system_exposed_libraries: libvraudio_client.qti.so:libbinauralrenderer_wrapper.qti.so:libhoaeffects.qti.so:libQOC.qti.so:libcamera_algoup_jni.xiaomi.so:libcamera_mianode_jni.xiaomi.so:libupdateprof.qti.so:libthermalclient.qti.so:libQOC.qti.so:libdiag_system.qti.so:libqape.qti.so:libqesdk_ndk_platform.qti.so:liblistenjni.qti.so
I/Perf: Connecting to perf service.
V/GraphicsEnvironment: ANGLE Developer option for 'com.jiangdg.ausbc' set to: 'default'
V/GraphicsEnvironment: Updatable production driver is not supported on the device.
I/ForceDarkHelperStubImpl: initialize for com.jiangdg.ausbc , ForceDarkOrigin
D/OpenGLRenderer: JNI_OnLoad success
I/MiuiForceDarkConfig: setConfig density:2.750000, mainRule:0, secondaryRule:0, tertiaryRule:0
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
I/MultiDex: VM with version 2.1.0 has multidex support
I/MultiDex: Installing application
[... cut ...]
I/OpenGLRenderer: Davey! duration=9223371843489ms; Flags=1, FrameTimelineVsyncId=21618, IntendedVsync=193364846222, Vsync=193364846222, InputEventId=0, HandleInputStart=193372193205, AnimationStart=193372196486, PerformTraversalsStart=193372197893, DrawStart=193390882841, FrameDeadline=193380512888, FrameInterval=193372177528, FrameStartTime=8333333, SyncQueued=193392482164, SyncStart=193392545601, IssueDrawCommandsStart=193392625184, SwapBuffers=193393566330, FrameCompleted=9223372036854775807, DequeueBufferDuration=434218, QueueBufferDuration=1010417, GpuCompleted=9223372036854775807, SwapBuffersCompleted=193394771278, DisplayPresentTime=446271846283262213, 
`

steagne, 2023-02-13:
[logcat.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/10723241/logcat.txt)
This is from 3.3.3

jiangdongguo, 2023-02-14:
In facet, there is no different between with them.so, how about branch feature_3.3.4 and it's log if it doesn't work.

steagne, 2023-02-14:
With feature_3.3.4 i cannot compile (i also tried updating Gradle to 7.6, but it didn't work). Two errors appear when building:
1) [stacktrace.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/10730624/stacktrace.txt)
2) CreateProcess error=2, Impossibile trovare il file specificato
(in english i guess it would have been "cannot locate specified file")

jiangdongguo, 2023-02-14:
Can you show me your 3.2.8 logs？

steagne, 2023-02-14:
Of course! Just one note: at the first opening, even if i give camera permission, it shows nothing. Then when i open the app again it works, so i give you both the logs.
First opening (fail): [AUSBC_v3.2.8_first.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/10734686/AUSBC_v3.2.8_first.txt)
Second open (success): [AUSBC_v3.2.8_second.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/10734687/AUSBC_v3.2.8_second.txt)

jiangdongguo, 2023-02-15:
Please update to new on branch master, if it doesn't work, paste me the log file.

steagne, 2023-02-15:
Still not working.. here's the log: [AUSBC_v3.3.3.log](https://github.com/jiangdongguo/AndroidUSBCamera/files/10741650/AUSBC_v3.3.3.log)
This time i've noticed on the log that now it shows available resolutions, but it seems to fail when trying to set aspect ratio

jiangdongguo, 2023-02-15:
update master and try again

steagne, 2023-02-15:
Wow great! Now it shows my camera! Really great work :)
The only flaw now, it shows a toast with an error message which i guess it's caused by the app trying to load last photo taken:
[image]

jiangdongguo, 2023-02-15:
yes,but it doesn't matter.

FriedrichM, 2023-02-21:
for me the demo app works with the current master that i built locally (linter was throwing errors but i fixed them)
implementation files('.../libausbc-release.aar') 
implementation files('.../libuvc-release.aar')
implementation files('.../libutils-release.aar')
implementation files('.../libuvccommon-release.aar')
but not when using 
implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.3.3'
(Unsupported preview size)

steagne, 2023-02-21:
Use:
implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:master-SNAPSHOT'

FriedrichM, 2023-02-21:
> Use: implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:master-SNAPSHOT'

this one works too Ill just wait for the next release then :)