jiangdongguo/AndroidUSBCamera #526: setPreviewSize failed

idan123456, 2023-02-15:
Hey, 
I'm trying when I try to connect one of my cameras to the device I'm getting the setPreviewSize failed error, but I try to set it to the preview size that the camera provides and I can understand why it gives me this error.
did anyone else also encounter it and knows how I can fix it?

Here is the Log file:
[LogThermal.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/10740551/LogThermal.txt)

jiangdongguo, 2023-02-15:
Update to new with branch master and have try.

jiangdongguo, 2023-02-15:
It's not the same. #522 cound not get the preview size list.

idan123456, 2023-02-15:
> It's not the same. #522 cound not get the preview size list.

I mean regarding complaining the master branch we both encountered :
CreateProcess error=2, The system cannot find the file specified

steagne, 2023-02-15:
> > It's not the same. #522 cound not get the preview size list.
> 
> I mean regarding complaining the master branch we both encountered : CreateProcess error=2, The system cannot find the file specified

Hi, i solved the problem by downloading NDK! File -> Project structure -> SDK Location -> Under "Android NDK location" press on "Download"

idan123456, 2023-02-15:
> Update to new with branch master and have a try.

With master branch, it does the same 
[LogMasterBranch.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/10744302/LogMasterBranch.txt)

idan123456, 2023-02-16:
@jiangdongguo 
When I try to debug the specific code that caused that error I get a different error, I can't understand what caused it but maybe you can help me with that please

[LogThermalDebug.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/10758242/LogThermalDebug.txt)

jiangdongguo, 2023-02-17:
how about version 3.2.8?

idan123456, 2023-02-21:
> how about version 3.2.8?

3.2.8 returns empty array on getAllPreviewSizes

ConnorFlis, 2023-12-18:
Has this been resolved? Any tips?