jiangdongguo/AndroidUSBCamera #528: Not seeing a preview

amglol, 2023-02-16:
I followed the README and my java class (Activity) extends the CameraActivity class. I implemented all of the required functions and I am not seeing any preview. Below is a the class I'm referring to. Any idea what I need to fix?

public class ExternalCameraActivity extends CameraActivity implements View.OnClickListener {
    ActivityExternalCameraBinding m_viewBinding;
    private final CaptureMediaView.CaptureMode mCameraMode = CaptureMediaView.CaptureMode.MODE_CAPTURE_PIC;

    public void onCameraBtnClick(View view) {
        switch (view.getId()) {
            case R.id.captureBtn:
                getImage();
        }
    }

    private void getImage() {
    }

    @Override
    protected void onCreate(@Nullable Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_external_camera);
    }

    private void handleCameraError(String msg) {
//        m_viewBinding.camera_view_container.setVisibility(View.VISIBLE);
//        m_viewBinding.frameRateTv.visibility = View.GONE;
    }

    private void handleCameraClosed() {
//        m_viewBinding.previewImg.setVisibility(View.VISIBLE);
//        m_viewBinding.frameRateTv.visibility = View.GONE;
    }

    private void handleCameraOpened() {
//        m_viewBinding.previewImg.setVisibility(View.GONE);
//        m_viewBinding.frameRateTv.visibility = View.VISIBLE;
    }

    @Nullable
    @Override
    protected View getRootView(@NonNull LayoutInflater layoutInflater) {
        if (m_viewBinding == null) {
            m_viewBinding = ActivityExternalCameraBinding.inflate(layoutInflater);
        }
        return m_viewBinding.getRoot();
    }

    @Nullable
    @Override
    protected IAspectRatio getCameraView() {
//        return new AspectRatioTextureView(getBaseContext());
        return null;
    }

    @Nullable
    @Override
    protected ViewGroup getCameraViewContainer() {
        return m_viewBinding.cameraViewContainer;
    }

    @Override
    public void onCameraState(@NonNull MultiCameraClient.ICamera iCamera, @NonNull State state, @Nullable String s) {
        switch (state) {
            case OPENED:
                handleCameraOpened();
            case CLOSED:
                handleCameraClosed();
            case ERROR:
                handleCameraError(s);
        }
    }

    @Override
    public int getGravity() { return Gravity.TOP; }

    @Override
    public UsbDevice getDefaultCamera() {
        return super.getDefaultCamera();
    }

    @Override
    public void onClick(View v) {

    }
}

jiangdongguo, 2023-02-17:
how about the demo shows?

amglol, 2023-02-17:
@jiangdongguo the demo worked on version 3.2.10, I was able to launch the application and my external camera was recognized and I could see a preview / capture image. Now with the latest tag '3.3.3' the build is failing

jiangdongguo, 2023-02-17:
Can you show the error message of 3.3.3 when you build?

amglol, 2023-02-20:
[image]

amglol, 2023-02-20:
For some reason building the project in Android Studio generates that error. However, when I downloaded the apk file onto my android phone the application works and I can see a preview / capture an image.

The problem I am still having is that I have added the following to my project and implemented the CameraActivity into my activity and I cannot see preview. The onCameraState function does get called and it generates an Error: null message and then the camera OPENED enum gets called but no preview. Any idea on what to try next?
    implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.3.3'
    implementation 'com.github.jiangdongguo.AndroidUSBCamera:libuvc:3.3.3'