jiangdongguo/AndroidUSBCamera #538: Video Feed Not Displaying Correctly

zachary-w-msi, 2023-03-02:
Hello,

I'm trying to test this library by building the sample application and I'm seeing weird behavior on the video feed. It seems to be the right data but the colors are incorrect. Is there a setting I need to change for this? Screenshot below. Thanks!
[image]

Drive Link with video and log files https://drive.google.com/drive/folders/1lNwknMupm7wFfQUb-RpVtx9k5wsvGzqY?usp=sharing

jiangdongguo, 2023-03-17:
How about closing opengl render？

zachary-w-msi, 2023-03-19:
I've tried setting RenderMode to NORMAL and I still see the same behavior. Is there another method to close the opengl render?

zachary-w-msi, 2023-03-22:
Also noticing a lot of E/BLASTBufferQueue: [VRI[MainActivity]#1](f:0,a:3) Faking releaseBufferCallback from transactionCompleteCallback in Logcat.

zachary-w-msi, 2023-03-27:
Closing issue seemed to be an issue with the UVC device. It's working now. Thanks @jiangdongguo

closed 2023-03-27 by zachary-w-msi