jiangdongguo/AndroidUSBCamera #539: 使用Device mic方式 ,报错无声

DaveZF, 2023-03-06:
版本：**V3.3.3**
问题：使用Demo，播放音频，使用AudioStrategyUAC方式报错，循环打印**“Audio loop: ----> 0”**

报错LOG：
2023-03-02 19:26:21.550 11614-11710 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 60 fps-->gl_render
2023-03-02 19:26:22.550 11614-11710 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 60 fps-->gl_render
2023-03-02 19:26:23.563 11614-11710 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 60 fps-->gl_render
2023-03-02 19:26:24.023 11614-11614 CAR_AUDIO               com.jiangdg.ausbc                    E  AudioManager playSoundEffectVolume!
2023-03-02 19:26:24.181 11614-11772 AudioThread             com.jiangdg.ausbc                    I  getHandler: 
2023-03-02 19:26:24.182 11614-11773 AudioThread             com.jiangdg.ausbc                    I  Audio thread start
2023-03-02 19:26:24.183 11614-11772 AUSBC                   com.jiangdg.ausbc                    I  [AudioUac]  initAudioRecord
2023-03-02 19:26:24.183 11614-11772 AUSBC                   com.jiangdg.ausbc                    I  [AudioUac]  startRecording:
2023-03-02 19:26:24.568 11614-11710 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 61 fps-->gl_render
2023-03-02 19:26:25.042 11614-11614 Choreographer           com.jiangdg.ausbc                    W  Frame time is 0.119059 ms in the future!  Check that graphics HAL is generating vsync timestamps using the correct timebase.
2023-03-02 19:26:25.572 11614-11710 AUSBC                   com.jiangdg.ausbc                    I  [RenderManager]  camera render frame rate is 60 fps-->gl_render
2023-03-02 19:26:25.688 11614-11773 USBAudio                com.jiangdg.ausbc                    D  before strdup vid:21325 pid:8457
2023-03-02 19:26:25.688 11614-11773 USBAudio                com.jiangdg.ausbc                    D  before libusb_init2 ret:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
2023-03-02 19:26:25.688 11614-11773 libusb/core             com.jiangdg.ausbc                    I  [11773*core.c:2085:libusb_init2]:libusb v1.0.19.10903
2023-03-02 19:26:25.688 11614-11773 libusb/usbfs            com.jiangdg.ausbc                    I  [11773*android_usbfs.c:512:op_init2]:call android_scan_devices
2023-03-02 19:26:25.688 11614-11773 USBAudio                com.jiangdg.ausbc                    D  before libusb_get_device_with_fd vid:21325 pid:8457
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D  open device err LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D  before scan_audio_interface ret:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D  scan_audio_interface
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D  scan_audio_interface :1
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D  scan_audio_interface :2
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D  set _speakerInterface 
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D  Get desc rate 96000
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D   _speakerInterface 3 _controlInterface 2 mMaxPacketSize 256 _alternateSetting 1
2023-03-02 19:26:25.689 11614-11773 USBAudio                com.jiangdg.ausbc                    D  Kernel Driver Active
2023-03-02 19:26:25.690 11614-11773 USBAudio                com.jiangdg.ausbc                    D  Kernel Driver Detached!
2023-03-02 19:26:25.690 11614-11773 USBAudio                com.jiangdg.ausbc                    D  kernel detach interface_number:3
2023-03-02 19:26:25.691 11614-11773 USBAudio                com.jiangdg.ausbc                    D  claim_interface r:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
2023-03-02 19:26:25.691 11614-11773 USBAudio                com.jiangdg.ausbc                    D  claim_interface r:0
2023-03-02 19:26:25.691 11614-11773 USBAudio                com.jiangdg.ausbc                    D  Select the  bEndpointAddress:130
2023-03-02 19:26:25.691 11614-11773 USBAudio                com.jiangdg.ausbc                    D  Select the altsetting:1, _speakerInterface:3
2023-03-02 19:26:25.698 11614-11773 USBAudio                com.jiangdg.ausbc                    D  set mic config fail 0 libusb:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED,errno:Socket operation on non-socket
2023-03-02 19:26:25.698 11614-11773 USBAudio                com.jiangdg.ausbc                    D  before sample_rate:48000
2023-03-02 19:26:25.698 11614-11773 USBAudio                com.jiangdg.ausbc                    D  host rate is 48000 ,device rate is 48000
2023-03-02 19:26:25.698 11614-11773 USBAudio                com.jiangdg.ausbc                    D  before interface_claim_if ret:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
2023-03-02 19:26:25.698 11614-11773 USBAudio                com.jiangdg.ausbc                    D  Set up the transfers
2023-03-02 19:26:25.698 11614-11773 USBAudio                com.jiangdg.ausbc                    D  before fill EndpointAddress:130, per_packet:256, packets:10, total_transfer_size:2560
2023-03-02 19:26:25.699 11614-11773 UACAudio                com.jiangdg.ausbc                    I  initAudio: 0, mNativePtr = 529837431840
2023-03-02 19:26:25.699 11614-11771 AUSBC                   com.jiangdg.ausbc                    I  [AACEncodeProcessor]  initAudioTrack: sample=-1,format=2,channelConfig=12,count=1
2023-03-02 19:26:25.700 11614-11771 android.me...AudioTrack com.jiangdg.ausbc                    E  getMinBufferSize(): -1 Hz is not a supported sample rate.
2023-03-02 19:26:25.700 11614-11773 AudioThread             com.jiangdg.ausbc                    I  handleInitAudioRecord
2023-03-02 19:26:25.700 11614-11773 UACAudio                com.jiangdg.ausbc                    I  startRecording: success
2023-03-02 19:26:25.700 11614-11773 AudioThread             com.jiangdg.ausbc                    I  handleStartRecording
2023-03-02 19:26:25.701 11614-11779 USBAudio                com.jiangdg.ausbc                    D  Audio loop: startCapture status=1, audioObject=0x2676
2023-03-02 19:26:25.701 11614-11779 USBAudio                com.jiangdg.ausbc                    D  Audio loop:
2023-03-02 19:26:25.702 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  [AACEncodeProcessor]  start/stop play mic failed, err = Invalid sample rate -1
2023-03-02 19:26:25.702 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  java.lang.IllegalArgumentException: Invalid sample rate -1
2023-03-02 19:26:25.702 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  	at android.media.AudioFormat$Builder.setSampleRate(AudioFormat.java:1050)
2023-03-02 19:26:25.702 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  	at android.media.AudioTrack.<init>(AudioTrack.java:697)
2023-03-02 19:26:25.702 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  	at android.media.AudioTrack.<init>(AudioTrack.java:632)
2023-03-02 19:26:25.702 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  	at com.jiangdg.ausbc.encode.AACEncodeProcessor.initAudioTrack(AACEncodeProcessor.kt:341)
2023-03-02 19:26:25.702 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  	at com.jiangdg.ausbc.encode.AACEncodeProcessor.playAudioStart$lambda-6(AACEncodeProcessor.kt:158)
2023-03-02 19:26:25.702 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  	at com.jiangdg.ausbc.encode.AACEncodeProcessor.lambda$K0Aqjmwt71ROgJwG6e0PYF8Kj6k(Unknown Source:0)
2023-03-02 19:26:25.703 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  	at com.jiangdg.ausbc.encode.-$$Lambda$AACEncodeProcessor$K0Aqjmwt71ROgJwG6e0PYF8Kj6k.run(Unknown Source:4)
2023-03-02 19:26:25.703 11614-11771 AUSBC                   com.jiangdg.ausbc                    E  	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
[... cut ...]
2023-03-02 19:26:25.780 11614-11779 USBAudio                com.jiangdg.ausbc                    D  Audio loop: ----> 0
2023-03-02 19:26:25.781 11614-11779 USBAudio                com.jiangdg.ausbc                    D  Audio loop:
2023-03-02 19:26:25.791 11614-11779 USBAudio                com.jiangdg.ausbc                    D  Audio loop: ----> 0
2023-03-02 19:26:25.791 11614-11779 USBAudio                com.jiangdg.ausbc                    D  Audio loop:
2023-03-02 19:26:25.799 11614-11779 USBAudio                com.jiangdg.ausbc                    D  Audio loop: ----> 0

jiangdongguo, 2023-03-17:
是录音还是实时播放？

DaveZF, 2023-03-29:
> 实时播放过程

zhudaoshun, 2023-09-23:
最新版3.3.3版本 实时播放没有声音 只有画面 怎么回事？