jiangdongguo/AndroidUSBCamera #540: UsbManager.openDevice Crash

toni1727, 2023-03-08:
Thank you very much for this wonderful library.

I just got this exception when disconnecting the camera:

E/UsbManager: exception in UsbManager.openDevice
    java.lang.IllegalArgumentException: device /dev/bus/usb/001/023 does not exist or is restricted
        at android.os.Parcel.createException(Parcel.java:1954)
        at android.os.Parcel.readException(Parcel.java:1918)
        at android.os.Parcel.readException(Parcel.java:1868)
        at android.hardware.usb.IUsbManager$Stub$Proxy.openDevice(IUsbManager.java:484)

Best regards.

jiangdongguo, 2023-03-17:
It looks like the device was pulled out. How it comes out？

dotSynergy, 2023-03-17:
Happening for me too, but it crashes just after the popup to confirm access.

`I/CameraUvc: attach device = UsbDevice[mName=/dev/bus/usb/001/004,mVendorId=1234,mProductId=5679,mVendorId=0x4d2,mProductId=0x162f,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=Rokid Corporation Ltd.,mProductName=Rokid Air Pro (5.0.1),mVersion=2.00,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@be4e227,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=192,mMaxPower=50,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=1,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=1,mAlternateSetting=1,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=9,mMaxPacketSize=68,mInterval=1]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=null,mClass=3,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=3,mMaxPacketSize=64,mInterval=1]
    UsbEndpoint[mAddress=2,mAttributes=3,mMaxPacketSize=64,mInterval=1]]
    UsbInterface[mId=3,mAlternateSetting=0,mName=null,mClass=3,mSubclass=1,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=64,mInterval=1]
    UsbEndpoint[mAddress=1,mAttributes=3,mMaxPacketSize=64,mInterval=1]]]]
I/CameraUvc: attach device = UsbDevice[mName=/dev/bus/usb/001/009,mVendorId=4726,mProductId=39973,mVendorId=0x1276,mProductId=0x9c25,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=ROKIDCAM-B706,mProductName=ROKIDCAM-B706,mVersion=72.08,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@e865cd4,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=250,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=ROKIDCAM-B706,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=135,mAttributes=3,mMaxPacketSize=16,mInterval=8]]
    UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=1,mAlternateSetting=1,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=192,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=2,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=384,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=3,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=512,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=4,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=640,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=5,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=800,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=6,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=944,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=7,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=2688,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=8,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=2848,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=9,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=3040,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=10,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=5056,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=11,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=5116,mInterval=1]]]]
D/MainActivity: onPause, true
W/USBMonitor: openDevice failed in UsbControlBlock11, wait and try again
D/MainActivity: onResume, true
I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@bf16cd8 time:157788179
D/MainActivity: onPause, true
E/UsbManager: exception in UsbManager.openDevice
    java.lang.IllegalArgumentException: device /dev/bus/usb/001/009 does not exist or is restricted
        at android.os.Parcel.createException(Parcel.java:2075)
        at android.os.Parcel.readException(Parcel.java:2039)
        at android.os.Parcel.readException(Parcel.java:1987)
        at android.hardware.usb.IUsbManager$Stub$Proxy.openDevice(IUsbManager.java:739)
        at android.hardware.usb.UsbManager.openDevice(UsbManager.java:449)
        at com.jiangdg.usb.USBMonitor$UsbControlBlock.<init>(USBMonitor.java:1024)
        at com.jiangdg.usb.USBMonitor$UsbControlBlock.<init>(USBMonitor.java:998)
        at com.jiangdg.usb.USBMonitor.lambda$processConnect$0$com-jiangdg-usb-USBMonitor(USBMonitor.java:597)
        at com.jiangdg.usb.USBMonitor$$ExternalSyntheticLambda0.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
E/AndroidRuntime: FATAL EXCEPTION: USBMonitor
    Process: com.smartremotesolutions.glasses, PID: 1588
[... cut ...]
        at com.jiangdg.usb.USBMonitor$UsbControlBlock.<init>(USBMonitor.java:998)
        at com.jiangdg.usb.USBMonitor.lambda$processConnect$0$com-jiangdg-usb-USBMonitor(USBMonitor.java:597)
        at com.jiangdg.usb.USBMonitor$$ExternalSyntheticLambda0.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
`