jiangdongguo/AndroidUSBCamera #546: grab frames for opencv

molisorme, 2023-03-26:
hi i read issues about how to use `addPreviewDataCallBack` and use `onPreviewData` to grab frames byte array but cant figure out how to make it work. can you provide a simple code example in Fragment or Activity?

jiangdongguo, 2023-03-28:
`public class UVCFragment extends MultiCameraFragment implements ICameraStateCallBack, IStateCallback, IPreviewDataCallBack{   
 @Override
    protected void onCameraConnected(@NonNull MultiCameraClient.ICamera camera) {
        Logger.INSTANCE.i(TAG, "camera connected, id = " + camera.getUsbDevice().getDeviceName());
        camera.openCamera(mViewBinding.tvCameraRender, getCameraRequest());
        camera.setPreviewDataCallBack(this);
        camera.setCameraStateCallBack(this);
        AusbcPusher.getInstance().init(getContext(), getAusbcConfig(camera), this);

    }

    @Override
    protected void onCameraDisConnected(@NonNull MultiCameraClient.ICamera camera) {
        Logger.INSTANCE.i(TAG, "camera connected, id = " + camera.getUsbDevice().getDeviceId());
        camera.closeCamera();
        AusbcPusher.getInstance().destroy();
    }

    @Override
    protected void onCameraAttached(@NonNull MultiCameraClient.ICamera camera) {
        Logger.INSTANCE.i(TAG, "camera attached, id = " + camera.getUsbDevice().toString());
    }

    @Override
    protected void onCameraDetached(@NonNull MultiCameraClient.ICamera camera) {
        Logger.INSTANCE.i(TAG, "camera detached, id = " + camera.getUsbDevice().getDeviceId());
    }

    private CameraRequest getCameraRequest() {
        return new CameraRequest.Builder()
                .setPreviewWidth(1280)
                .setPreviewHeight(720)
                .create();
    }

    @Override
    public void onCameraState(@NonNull MultiCameraClient.ICamera camera, @NonNull State code, @Nullable String msg) {
        switch (code) {
            case OPENED:
                isCameraOpened = true;
                handleCameraOpened();
                break;
            case CLOSED:
                isCameraOpened = false;
                handleCameraClosed();
                break;
            case ERROR:
                isCameraOpened = false;
                ToastUtils.show(TextUtils.isEmpty(msg) ? "未知异常" : Objects.requireNonNull(msg));
                break;
        }
    }
    @Override
    public void onPreviewData(@Nullable byte[] data, @NonNull DataFormat format) {
        if (data == null) {
            return;
        }
        AusbcPusher.getInstance().pushStream(1, data, data.length, System.currentTimeMillis() * 1000);
    }
}`

openmotion, 2023-06-14:
do you have a kotlin version, in a fragment ?
thanks
because for me addPreviewDataCallBack is not fired