jiangdongguo/AndroidUSBCamera #559: Android 12 not working

dgjinovci, 2023-04-28:
does not work in Android 12 (level 31)

Is there any update for this version?

ColtonIdle, 2023-05-05:
I think i was using it with android 12 and 13 devices without an issue. Do you have a stacktrace @dgjinovci ?

dgjinovci, 2023-05-06:
@ColtonIdle unfortunately I do not have stacktrace, we've changed the tablet with Android 13 (level 33) and it works fine 
Guess this is a device specific error and not Android version 

If running a Lenovo Tab M10 Pro (Gen 3) & it was a signal 11 (SIGSEGV), code 2 (SEGV_ACCERR) in libuvc.so

Thought this was the issue [https://developer.android.com/about/versions/10/behavior-changes-all#xom-binaries](url)
and did set LOCAL_XOM := false on all Android.mk included in libuvc/src/main/jni/Android.mk but no luck