jiangdongguo/AndroidUSBCamera #563: 无法打开 不支持预览尺寸

wsdydeni, 2023-05-24:
```
 aspect ratio = null, getAllPreviewSizes = [], 
open camera failed, preview size(PreviewSize(width=640, height=480)) unsupported-> []

    private fun getCameraRequest(): CameraRequest {
        return CameraRequest.Builder()
            .setPreviewWidth(340)
            .setPreviewHeight(280)
            .create()
    }
```
设置的预览无效啊

Cyril-Lee, 2023-06-13:
建議先用 getSuitableSize 取得 size, 再用 openCamera 開啟, getAllPreviewSizes 要成功開啟才會有資料

LiamDai, 2024-01-29:
遇到了一摸一样的错误，你们是怎么解决的？

[CameraUVC]  open camera failed.
java.lang.UnsupportedOperationException: open failed:result=-99
[CameraUVC]  open camera failed, preview size(PreviewSize(width=640, height=480)) unsupported-> null
[SurfaceTexture-0-26548-0](id:67b400000000,api:0,p:-1,c:26548) disconnect: not connected (req=1)

zhouqunhuang, 2024-01-30:
我也是这个问题！你们怎么解决了吗？困扰几天都没办法解决。请求帮忙

ethaannn, 2024-04-27:
再3.3.3 版本中我是这样解决预览问题的 
 override fun generateCamera(ctx: Context, device: UsbDevice): MultiCameraClient.ICamera  {
        val  c =CameraUVC(ctx, device)
        c.isPreviewSizeSupported(PreviewSize(ctx.dp2px(500),ctx.dp2px(500)))
        return c
    }

ethaannn, 2024-04-27:
> 再3.3.3 版本中我是这样解决预览问题的 override fun generateCamera(ctx: Context, device: UsbDevice): MultiCameraClient.ICamera { val c =CameraUVC(ctx, device) c.isPreviewSizeSupported(PreviewSize(ctx.dp2px(500),ctx.dp2px(500))) return c }

 private  fun getCameraRequest(): CameraRequest {
        return CameraRequest.Builder()
            .setPreviewHeight(activity?.baseContext?.dp2px(500)!!)
            .setPreviewWidth(activity?.baseContext?.dp2px(500)!!)
            .create()
    }