jiangdongguo/AndroidUSBCamera #564: getCameraView()

todaycodemaster, 2023-05-25:
override fun getCameraView(): IAspectRatio {
      return AspectRatioTextureView(requireContext())
}

As I think this function is not working. 

I connect webcam and webcam is working well in emulator default camera.

But this app is showing black screen.

Please let me know what is the problem

I read your read me file but I can't follow it.
[image]

todaycodemaster, 2023-05-25:
I found the reason. My device is android 13.

It is not working on android 13 devices 

Motorola Moto G Stylus
Samsung A14 5G
Samsung S22

Camera is connected successfully.

[image]

But as you can see, it is not working.
[image]

How can I solve this problem?

Nalethen, 2023-06-14:
I have had this problem also and did some investigation into it. It seems that it is not caused by Android 13 since in some phones with it, like Samsung S21, it works. The phones I tested which didn’t work were Samsung A12, A13 and A33. I’m also pretty sure it happened on some of these back when they still had Android 12. Also one USB camera worked if I selected the lowest resolution it had (160x120) but anything higher and it didn’t work.

I managed to get it working by editing [libuvc_internal.h](https://github.com/jiangdongguo/AndroidUSBCamera/blob/c459efc340e20faa34235c20710e5df9c54eb7cd/libuvc/src/main/jni/libuvc/include/libuvc/libuvc_internal.h#L250) and modifying [LIBUVC_NUM_TRANSFER_BUFS 10](https://github.com/jiangdongguo/AndroidUSBCamera/blob/c459efc340e20faa34235c20710e5df9c54eb7cd/libuvc/src/main/jni/libuvc/include/libuvc/libuvc_internal.h#L250) to be smaller. 7 was the highest where it still worked for me so I left it at 5. 

This seems like just a workaround though and that the root problem is somewhere else. The problem seemed to be that frame buffers in libuvc were getting corrupted, possibly because the transfer buffers were getting overwritten or cleared before the data had been fully processed by the consuming side. When requesting mjpeg this results in decoding error and no frames will be produced, but when requesting YUV the corrupted frames will be shown on the preview. It looks to me like some concurrency or timing issue, but I don’t have much experience with C language or USB protocols so trying to debug the problem has been pretty challenging. For now I will just go with lowering the LIBUVC_NUM_TRANSFER_BUFS and if no problems come up leave it at that.

todaycodemaster, 2023-06-16:
Thanks for your reply.

I updated as you guide but there is nothing difference.

Is there any way to solve this problem?

Saurabh-WowLabz-Kataria, 2023-07-10:
@onemetafox  did you find any solution?

onemetafox, 2023-07-10:
I didn't get solution yet.

mh-ahmadii, 2023-10-30:
Hey @Nalethen, I hope you're doing well.
I have a question: How do you change the value of [LIBUVC_NUM_TRANSFER_BUFS 10]?
I'm a bit confused about how to modify the native code of a third-party library because I add the library to my project through Gradle.Could you please help me understand how I can change this value?

Nalethen, 2023-10-30:
> Hey @Nalethen, I hope you're doing well. I have a question: How do you change the value of [LIBUVC_NUM_TRANSFER_BUFS 10]? I'm a bit confused about how to modify the native code of a third-party [...]

I cloned the repository and added the code to my project as modules. You can then modify the library code but it is not optimal adds its own problems. Might have to edit the gradle files to even get it to compile with your own project, I think the code wasn't compatible with the newest Android Gradle Plugin version for example. But I don't remember the steps I had to take to get it to work so well anymore.

Also in libuvc module after compiling I think it put the updated native libraries to "lib" directory but the app reads them from "jniLibs" directory, so I had to manually copy the **.so** files to "jniLibs" for the modified native code to actually be used by the app.

mh-ahmadii, 2023-10-31:
Thank you so much for your help and for sharing your experience.

I'm currently facing an issue with a black screen in my project, and I'm wondering if it could be related to the RAM value of the devices. What are your thoughts on this?
To modify that parameter, I forked the library on GitHub, made some changes, and published it on jitpack.io, subsequently implementing it in my project. However, when I set the LIBUVC_NUM_TRANSFER_BUFS value to 5, it doesn't seem to work for wick devices.
Could you please provide some guidance on how you resolved this problem by changing the LIBUVC_NUM_TRANSFER_BUFS value?

mh-ahmadii, 2023-10-31:
> I cloned the repository and added the code to my project as modules. You can then modify the library code but it is not optimal adds its own problems. Might have to edit the gradle files to even get [...]

Thank you so much for your help and for sharing your experience.

I'm currently facing an issue with a black screen in my project, and I'm wondering if it could be related to the RAM value of the devices. What are your thoughts on this?
To modify that parameter, I forked the library on GitHub, made some changes, and published it on jitpack.io, subsequently implementing it in my project. However, when I set the LIBUVC_NUM_TRANSFER_BUFS value to 5, it doesn't seem to work for wick devices.
Could you please provide some guidance on how you resolved this problem by changing the LIBUVC_NUM_TRANSFER_BUFS value?