jiangdongguo/AndroidUSBCamera #600: 多次尝试打开/关闭相机导致内存溢出，系统关机

makeking, 2023-08-31:
[system_server_lowmem@1693456176835.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/12483470/system_server_lowmem%401693456176835.txt)
[SYSTEM_TOMBSTONE@1693444874054.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/12483471/SYSTEM_TOMBSTONE%401693444874054.txt)
[SYSTEM_TOMBSTONE@1693448158454.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/12483475/SYSTEM_TOMBSTONE%401693448158454.txt)

makeking, 2023-08-31:
Build: Raindi/full_ap7350_65u_l1/ap7350_65u_l1:5.1/LMY47D/72fc93e:user/release-key
Hardware: unknown
Revision: 0
Bootloader: unknown
Radio: unknown
Kernel: Linux version 3.10.65 (wangwm@r410b) (gcc version 4.9.x-google 20140827 (mtk-20150409) (GCC) ) #1 SMP PREEMPT Tue Feb 28 21:22:01 CST 2023

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'Raindi/full_ap7350_65u_l1/ap7350_65u_l1:5.1/LMY47D/72fc93e:user/release-key'
Revision: '0'
ABI: 'arm'
pid: 3241, tid: 3463, name: Thread-70  >>> com.bete.cellcounter <<<
signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xddc02000
    r0 ddc02000  r1 db65cda0  r2 00000040  r3 ff010101
    r4 ff010101  r5 ff010101  r6 ff010101  r7 ff010101
    r8 ff010101  r9 db65c400  sl 00000a00  fp dbb01d60
    ip ff010101  sp dbb01cd8  lr ff010101  pc f70d16ec  cpsr 20070010
    d0  0000000000000000  d1  00000000ffffffff
    d2  0000000000000034  d3  0000000000000131
    d4  efc39b00efc39b00  d5  0000000000000000
    d6  0000000000002710  d7  7056f6887056f688
    d8  000000003f800000  d9  0000000000000000
    d10 0000000000000000  d11 0000000000000000
    d12 0000000000000000  d13 0000000000000000
    d14 0000000000000000  d15 0000000000000000
    d16 0000000000000001  d17 0000000000000001
    d18 0000000000000000  d19 0000000000000001
    d20 0009600000000000  d21 0000000000000bf4
    d22 0000000000000000  d23 0000000000000000
    d24 7056f6887056f688  d25 7056f6887056f688
    d26 7056f6887056f688  d27 7056f6887056f688
    d28 7056f6887056f688  d29 7056f6887056f688
    d30 7056f6887056f688  d31 7056f6887056f688
    scr 68000013

backtrace:
    #00 pc 000136ec  /system/lib/libc.so (__memcpy_base_aligned+60)
    #01 pc 00011b2c  /data/app/com.bete.cellcounter-2/lib/arm/libUVCCamera.so (UVCPreview::do_capture_surface(_JNIEnv*)+416)
    #02 pc 00011980  /data/app/com.bete.cellcounter-2/lib/arm/libUVCCamera.so (UVCPreview::do_capture(_JNIEnv*)+160)
    #03 pc 00011038  /data/app/com.bete.cellcounter-2/lib/arm/libUVCCamera.so (UVCPreview::capture_thread_func(void*)+64)
    #04 pc 00016feb  /system/lib/libc.so (__pthread_start(void*)+30)
    #05 pc 0001501f  /system/lib/libc.so (__start_thread+6)

stack:
         dbb01c98  00000002  
         dbb01c9c  00000010  
         dbb01ca0  dbb01db8  [stack:3463]
         dbb01ca4  f6e31e29  /system/lib/libgui.so (android::Surface::lock(ANativeWindow_Buffer*, ARect*))
         dbb01ca8  dbb01d14  [stack:3463]
         dbb01cac  de072df0  
         dbb01cb0  efca0624  
         dbb01cb4  efd43520  
         dbb01cb8  00000000  
         dbb01cbc  00000a00  
         dbb01cc0  dbb01d60  [stack:3463]
         dbb01cc4  f6e31513  /system/lib/libgui.so (android::Surface::dispatchLock(std::__va_list)+10)
         dbb01cc8  f6e30d83  /system/lib/libgui.so (android::Surface::hook_perform(ANativeWindow*, int, ...))
         dbb01ccc  f6e30d9b  /system/lib/libgui.so (android::Surface::hook_perform(ANativeWindow*, int, ...)+24)
         dbb01cd0  de14de08  
         dbb01cd4  dbb01ce0  [stack:3463]
    #00  dbb01cd8  00000a00  
         dbb01cdc  db65c400  
         dbb01ce0  00000000  
         dbb01ce4  00000002  
         dbb01ce8  00000b40  
         dbb01cec  ddc01680  anon_inode:dmabuf
         dbb01cf0  ddc01680  anon_inode:dmabuf
         dbb01cf4  df19ab30  /data/app/com.bete.cellcounter-2/lib/arm/libUVCCamera.so (UVCPreview::do_capture_surface(_JNIEnv*)+420)
    #01  dbb01cf8  efca0624  
         dbb01cfc  efd43520  
[... 1602 lines cut ...]
--- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
pid: 3241, tid: 3286, name: Binder_3  >>> com.bete.cellcounter <<<
cannot get registers: No such process

backtrace:
    #00 pc 0003b81c  /system/lib/libc.so (__ioctl+8)
    #01 pc 00053539  /system/lib/libc.so (ioctl+14)
    #02 pc 0001f3db  /system/lib/libbinder.so (android::IPCThreadState::talkWithDriver(bool)+190)
    #03 pc 0001f933  /system/lib/libbinder.so (android::IPCThreadState::getAndExecuteCommand()+6)
    #04 pc 0001f995  /system/lib/libbinder.so (android::IPCThreadState::joinThreadPool(bool)+48)
    #05 pc 00023d77  /system/lib/libbinder.so
    #06 pc 00010a51  /system/lib/libutils.so (android::Thread::_threadLoop(void*)+112)
    #07 pc 0006305d  /system/lib/libandroid_runtime.so (android::AndroidRuntime::javaThreadShell(void*)+72)
    #08 pc 000105bf  /system/lib/libutils.so
    #09 pc 00016feb  /system/lib/libc.so (__pthread_start(void*)+30)
    #10 pc 0001501f  /system/lib/libc.so (__start_thread+6)

stack:
         ........  ........
         ........  ........
         ........  ........
         ........  ........
         ........  ........
         ........  ........
         .......[[TRUNCATED]]

makeking, 2023-08-31:
经过查看会发生两种bug，signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), 与 signal 11 (SIGSEGV), code 1 (SEGV_ACCERR),查看问题发现是内存不足导致的，但是暂时不知道问题如何解决，大神请帮忙看一下

qihao3768, 2023-09-15:
我们也是这种类似问题，我们是MTK平台8735的板子 1+8的配置，怎么处理呢

kbxwn, 2023-09-18:
试下降低摄像头分辨率看看

alexluocj, 2024-10-09:
@makeking 请问你们这个问题解决了吗？

alexluocj, 2024-10-09:
> 经过查看会发生两种bug，signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), 与 signal 11 (SIGSEGV), code 1 (SEGV_ACCERR),查看问题发现是内存不足导致的，但是暂时不知道问题如何解决，大神请帮忙看一下

请问你们怎么解决的？