jiangdongguo/AndroidUSBCamera #606: Audio Mic issue

bhumikacodezee, 2023-10-03:
startPlayMic method not working in latest version(Audio not play)

danji, 2023-11-24:
App will crash after usb plugged-out when playing audio.

== tombstone_22
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'Lenovo/LenovoYT-K606F/K606F:13/TKQ1.221013.002/S510140_230926_ROW:user/release-keys'
Revision: '0'
ABI: 'arm64'
Timestamp: 2023-11-24 15:36:03.310728236+0800
Process uptime: 25s
Cmdline: com.jiangdg.ausbc
pid: 12127, tid: 12137, name: HeapTaskDaemon  >>> com.jiangdg.ausbc <<<
uid: 10169
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: 'terminating with uncaught exception of type NSt3__112system_errorE: mutex lock failed: Device or resource busy'
    x0  0000000000000000  x1  0000000000002f69  x2  0000000000000006  x3  0000007494bf6f60
    x4  0000000000000010  x5  0000000000000010  x6  0000000000000010  x7  7f7f7f7f7f7f7f7f
    x8  00000000000000f0  x9  00000078396a3b70  x10 0000000000000001  x11 00000078396e378c
    x12 0000000000789458  x13 00000175d7a01ffd  x14 000e9942b7e471e8  x15 0000000034155555
    x16 000000783974bd70  x17 0000007839727420  x18 00000074942b0000  x19 0000000000002f5f
    x20 0000000000002f69  x21 00000000ffffffff  x22 0000007494bf7090  x23 ffffff80ffffffc8
    x24 000000783974df90  x25 0000000000000000  x26 b4000076455d6010  x27 000000783619c2ac
    x28 0000007494bf8000  x29 0000007494bf6fe0
    lr  00000078396d46c8  sp  0000007494bf6f40  pc  00000078396d46f4  pst 0000000000001000

bugreport as attached file

[bugreport-LenovoYT-K606F-TKQ1.221013.002-2023-11-24-15-36-12.zip](https://github.com/jiangdongguo/AndroidUSBCamera/files/13456424/bugreport-LenovoYT-K606F-TKQ1.221013.002-2023-11-24-15-36-12.zip)

jenny850113, 2025-01-03:
> 我透過修改mAudioTrack初始化參數可以正常播放，但播放時拔出usb程式崩潰
請問是做了哪些的初始化呢?

jenny850113, 2025-01-03:
> 我透過修改mAudioTrack初始化參數可以正常播放，但播放時拔出usb程式崩潰

請問是做了哪些初始化呢?