jiangdongguo/AndroidUSBCamera #620: 连续开关摄像头2次以上，再次开启会失败

hjywyj, 2023-11-20:
open camera failed.
   java.lang.UnsupportedOperationException: open failed:result=-1----->id_camera=2267452288;venderId=;productId=;fileDescriptor=;busNum=;devAddr=;usbfs=
            Exception：com.jiangdg.usb.USBMonitor.access$1400(USBMonitor.java:59)
            com.jiangdg.usb.USBMonitor$UsbControlBlock.<init>(USBMonitor.java:1059)
            com.jiangdg.usb.USBMonitor$UsbControlBlock.clone(USBMonitor.java:1091)
            com.jiangdg.uvc.UVCCamera.open(UVCCamera.java:193)
            com.jiangdg.ausbc.camera.CameraUVC.openCameraInternal(CameraUVC.kt:125)
            com.jiangdg.ausbc.MultiCameraClient$ICamera$handleMessage$1$1.onSurfaceTextureAvailable(MultiCameraClient.kt:333)
            com.jiangdg.ausbc.render.RenderManager.startRenderScreen(RenderManager.kt:267)
            com.jiangdg.ausbc.MultiCameraClient$ICamera.handleMessage(MultiCameraClient.kt:326)
            android.os.Handler.dispatchMessage(Handler.java:98)
            android.os.Looper.loop(Looper.java:154)
            android.os.HandlerThread.run(HandlerThread.java:61)
            core message ->Attempt to read from field 'android.hardware.usb.UsbManager com.jiangdg.usb.USBMonitor.mUsbManager' on a null object reference
            	at com.jiangdg.uvc.UVCCamera.open(UVCCamera.java:217)
            	at com.jiangdg.ausbc.camera.CameraUVC.openCameraInternal(CameraUVC.kt:125)
            	at com.jiangdg.ausbc.MultiCameraClient$ICamera$handleMessage$1$1.onSurfaceTextureAvailable(MultiCameraClient.kt:333)
            	at com.jiangdg.ausbc.render.RenderManager.startRenderScreen(RenderManager.kt:267)
            	at com.jiangdg.ausbc.MultiCameraClient$ICamera.handleMessage(MultiCameraClient.kt:326)
            	at android.os.Handler.dispatchMessage(Handler.java:98)
            	at android.os.Looper.loop(Looper.java:154)
            	at android.os.HandlerThread.run(HandlerThread.java:61)
 open camera failed, preview size(PreviewSize(width=640, height=480)) unsupported-> null
 call to OpenGL ES API with no current context (logged once per thread)

hjywyj, 2023-11-21:
原因是UsbControlBlock的WeakReference<USBMonitor> mWeakMonitor和WeakReference<UsbDevice> mWeakDevice被回收掉了，导致open方法的 ctrlBlock.clone()复制了个null出来，然后空指针异常了

hjywyj, 2023-11-21:
解决方案在onConnectDev->setUsbControlBlock缓存下UsbControlBlock的usbMonitor，每次openCamera前重新调用setUsbControlBlock

CandyTek, 2024-02-21:
大佬，可以把onConnectDev 方法拿出来参考参考吗