jiangdongguo/AndroidUSBCamera #627: Black Screen After Format Change

RobbieElias, 2023-12-14:
Hello,

The app works great when first booted. But if I switch the video format (e.g. 1080p 30FPS YUV to 720p 30FPS YUV), then the preview turns black and no more frames come in. I have to unplug device to make it work again.

I am using a capture card based on the ms2130 chip. I have tested using the USB camera app on the Play Store and it works fine on there.

Appreciate any help!

RobbieElias, 2024-01-17:
Finally was able to figure it out. I think this could be helpful to some. Change the following lines in the `stream.c` file:

```
case LIBUSB_TRANSFER_ERROR:
	UVC_DEBUG("not retrying transfer, status = %d", transfer->status);
//	MARK("not retrying transfer, status = %d", transfer->status);
//	_uvc_delete_transfer(transfer);
	resubmit = 0;
	break;
```

To:

```
case LIBUSB_TRANSFER_ERROR:
	libusb_clear_halt(strmh->devh->usb_devh, strmh->stream_if->bEndpointAddress);
	UVC_DEBUG("not retrying transfer, status = %d", transfer->status);
//	MARK("not retrying transfer, status = %d", transfer->status);
//	_uvc_delete_transfer(transfer);
	resubmit = 0;
	break;
```

closed 2024-01-17 by RobbieElias

JeffZane, 2024-11-01:
> Finally was able to figure it out. I think this could be helpful to some. Change the following lines in the `stream.c` file:
>
> ```
> case LIBUSB_TRANSFER_ERROR:
> 	UVC_DEBUG("not retrying transfer, [...]

I have tried this code but it has no effect, any other updates?