jiangdongguo/AndroidUSBCamera #638: Camera is not working on above android 12

bhanu-mds, 2024-01-24:
I have tried to run camera on android 13/14 but only black screen is visible, no camera preview is available.
Checked on Pixel 7 & Xiaomi Redmi 10 prime
[image]

shyam311, 2024-01-25:
this is really buggy repository, can't use in production apps. I facing issue like Grey camera preview, it captures corrupted images.

dhameliyadharmesh, 2024-02-06:
@shyam311 @bhanu-mds , Any solution?

ronaldsampaio, 2024-02-08:
The image capturing problem I solved by using the implementation change on [this issue](https://github.com/jiangdongguo/AndroidUSBCamera/issues/618#issue-1988661720)

dhameliyadharmesh, 2024-02-08:
I'm not able to detect Camera Preview in Android 12 and 13. Do you have any solution?

ronaldsampaio, 2024-02-08:
> @ronaldsampaio I'm not able to detect Camera Preview in Android 12 and 13. Do you have any solution?

So this is a Composable Scaffold that I use to get the camera preview. My app targets Android 13
```kotlin
    Scaffold(modifier = Modifier.fillMaxSize()) {
        Column(
            modifier = Modifier.fillMaxSize(),
            horizontalAlignment = Alignment.CenterHorizontally,
            verticalArrangement = Arrangement.Center
        ) {
            Box(modifier = Modifier.height(200.dp))
            Box(Modifier.height(300.dp)) {
                AndroidView(
                    factory = {
                        AspectRatioTextureView(it).apply {
                            this.surfaceTextureListener = (object : TextureView.SurfaceTextureListener{
                                val _tag = "camera_streaming"
                                override fun onSurfaceTextureAvailable(
                                    surface: SurfaceTexture,
                                    width: Int,
                                    height: Int
                                ) {
                                    //this@apply.setSurfaceTexture(surface)
                                    cameraClient.openCamera(this@apply)
                                }

                                override fun onSurfaceTextureSizeChanged(
                                    surface: SurfaceTexture,
                                    width: Int,
                                    height: Int
                                ) {
                                    Log.d(_tag, "onSurfaceTextureSIZECHANGED")
                                    cameraClient.setRenderSize(width, height)
                                }

                                override fun onSurfaceTextureDestroyed(surface: SurfaceTexture): Boolean {
                                    Log.d(_tag, "onSurfaceTextureDESTROYED")
                                    cameraClient.closeCamera()
                                    return true
                                }

                                override fun onSurfaceTextureUpdated(surface: SurfaceTexture) {
                                    //Log.d(_tag, "onSurfaceTextureUPDATED")
                                }

                            })
                        }.also {
                            usbCameraViewModel.startVideoStreaming()

                        }
                    }
                )
//                AndroidView(
//                    factory = { ctx ->
//                        AspectRatioSurfaceView(ctx).apply {
//                            this.holder.addCallback(object : SurfaceHolder.Callback {
//                                override fun surfaceCreated(holder: SurfaceHolder) {
//                                    cameraClient.openCamera(this@apply)
//                                }
//
//                                override fun surfaceChanged(
//                                    holder: SurfaceHolder,
//                                    format: Int,
//                                    width: Int,
//                                    height: Int
//                                ) {
//                                    cameraClient.setRenderSize(width, height)
//                                }
//
//                                override fun surfaceDestroyed(holder: SurfaceHolder) {
//                                    cameraClient.closeCamera()
//                                }
//                            })
//                        }
//                    }
//                )
            }
```

There is two approaches in this code. First with TextureView(AspectRatioTextureView) and SurfaceView(AspectRatioSurfaceView)
I'm using the Texture view because it solved the black problem after the app went to background and back to foreground.

dhameliyadharmesh, 2024-02-10:
Android 12, I'm not able to detect the USB Camera device. Do you have solution for it?