jiangdongguo/AndroidUSBCamera #640: How can I get the camera preview data? (as a bitMap would be great)

runefor, 2024-01-30:
I tried to get the data with addPreviewDataCallBack inside a class that inherits from "CameraFragment()", but I'm not sure if the way I wrote it is correct.

`override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        getCurrentCamera()?.addPreviewDataCallBack(object: IPreviewDataCallBack{
            override fun onPreviewData(
                data: ByteArray?,
                width: Int,
                height: Int,
                format: IPreviewDataCallBack.DataFormat
            ) {
                Log.d("drawTest1", "onPreviewData: $data")
                if (data != null) {
                    val b: Bitmap?
                    //b = BitmapFactory.decodeByteArray(data, 0, data.size)
                    b = bitmapFromRgba(width, height, data)
                    Log.d("drawTest1", "onPreviewData: $b")
                }else{
                    Log.d("drawTest1", "No data?")
                }
            }
        })

        addPreviewDataCallBack(object : IPreviewDataCallBack{
            override fun onPreviewData(
                data: ByteArray?,
                width: Int,
                height: Int,
                format: IPreviewDataCallBack.DataFormat
            ) {
                Log.d("drawTest2", "onPreviewData: $data")
                if (data != null) {
                    val b: Bitmap?
                    //b = BitmapFactory.decodeByteArray(data, 0, data.size)
                    b = bitmapFromRgba(width, height, data)
                    Log.d("drawTest2", "onPreviewData: $b")
                }else{
                    Log.d("drawTest2", "No data?")
                }
            }
        })
    }`

bhanu-mds, 2024-01-30:
@runefor are you able to open camera in above android 12 ?

runefor, 2024-01-31:
> @runefor are you able to open camera in above android 12 ?

yes

AnyLifeZLB, 2025-02-10:
@runefor can you get a bitmap now ? addPreviewDataCallBack