jiangdongguo/AndroidUSBCamera #646: Cant find camera library

nsmith1024, 2024-03-05:
Im getting this error now when building in Android Studio. Seems it cant find this library used by the camera:

 com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.3.3

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:checkDebugAarMetadata'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.3.3.
     Searched in the following locations:
       - https://github.com/jitsi/jitsi-maven-repository/raw/master/releases/com/github/jiangdongguo/AndroidUSBCamera/libausbc/3.3.3/libausbc-3.3.3.pom
       - https://dl.google.com/dl/android/maven2/com/github/jiangdongguo/AndroidUSBCamera/libausbc/3.3.3/libausbc-3.3.3.pom
       - https://jcenter.bintray.com/com/github/jiangdongguo/AndroidUSBCamera/libausbc/3.3.3/libausbc-3.3.3.pom
     Required by:
         project :app > org.jitsi.react:jitsi-meet-sdk:9.0.0 > com.facebook.react:react-native-webrtc:118.0.2-jitsi-16324891

pelpanagiotis, 2024-03-05:
I am facing the same issue, as well

abrudana, 2024-09-11:
I am also facing the same issue. :(