jiangdongguo/AndroidUSBCamera #658: 如何解决分辨率大于1080P得时候画面延迟的问题。

zhouqunhuang, 2024-04-26:
当我设置高于2K画质的时候画面会有很大的延迟大概是2-3秒左右该如何修复这个问题。

jinyuer, 2024-05-28:
我也遇到这个问题

xiaohua717, 2024-07-22:
你们解决这个问题了吗？我找到是哪里导致的，是因为libjpeg-turbo把mjpeg转YUV的时候使用CPU复制数据造成，已确定这个库有使用SIMD技术，但我没有方案如何去再做优化。

Qdafengzi, 2025-03-18:
如何解决的贴一下代码啊 老铁

Qdafengzi, 2025-03-18:
@xiaohua717 
你解决了吗？

hiboxgit, 2025-03-18:
提示你一下，用并发去解决，它以前是串行转换的，如果分辨率高转换耗时长就会影响下一帧的转换

Qdafengzi, 2025-03-18:
> 提示你一下，用并发去解决，它以前是串行转换的，如果分辨率高转换耗时长就会影响下一帧的转换 [@Qdafengzi](https://github.com/Qdafengzi) ，代码我就不贴了

并发不会影响帧的顺序吧？

hiboxgit, 2025-03-18:
这个帧的顺序你自己去控制咯，大部份时间不会，偶尔是会的，你自己去通过代码保障

Qdafengzi, 2025-03-18:
@hiboxgit 大佬处理好了， 为何不开源，为开源事业做一份贡献[破涕为笑]

hiboxgit, 2025-03-18:
我不是大佬，所以我不要开源

Qdafengzi, 2025-03-18:
那可以单独给我一下吗？

hiboxgit, 2025-03-18:
我都提示你修改的方向了啊，还想不劳而获啊，付费

Qdafengzi, 2025-03-20:
@hiboxgit 你优化后可以达到4K 60帧吗， 或者非4K也可以60帧

hiboxgit, 2025-03-20:
> [@hiboxgit](https://github.com/hiboxgit) 你优化后可以达到4K 60帧吗， 或者非4K也可以60帧

首先，这个能不能60帧跟设备的性能有关系，第二个就是它代码库里面有一些限制帧率的地方，自己去找一下， 还有就是你接入的U VC摄像头本身是否支持输出高帧率， 测试的方式就是你先不进行 变换转换，直接循环读取他的视频帧，通过日志打印看看帧率够不够，如果帧率高的情况下再优化并发解码转换和注意0拷贝，还有GPU转换等措施，如果这些软件优化措施还不行，那就是设备本身芯片的性能跟不上了只能考虑换主控芯片了

hiboxgit, 2025-03-20:
> [@hiboxgit](https://github.com/hiboxgit) 你优化后可以达到4K 60帧吗， 或者非4K也可以60帧

想对标的话，你就直接参考沈垚的那个APP，他上架googleplay的，他那个优化的比较好了，延迟和帧率都比较理想，如果他的那个应用可以实现你的目的，起码就不是芯片的瓶颈，理论上你就是有机会通过软件优化达到他的那个app的水平的，或者你就直接用他的那个。

Qdafengzi, 2025-03-21:
@hiboxgit 感谢 按照你的思路 今天彻底解决了。

AndyYw, 2025-06-12:
@Qdafengzi 大佬， 怎么解决的，能指导一下不