jiangdongguo/AndroidUSBCamera #667: Android 14  Api 34崩溃了

darkSuperman, 2024-06-01:
java.lang.IllegalArgumentException:com.nextai.object: Targeting U+ (version 34 and above) disallows creating or retrieving a PendingIntent with FLAG_MUTABLE, an implicit Intent within and without FLAG_NO_CREATE and FLAG_ALLOW_UNSAFE_IMPLICIT_INTENT for security reasons. To retrieve an already existing PendingIntent, use FLAG_NO_CREATE, however, to create a new PendingIntent with an implicit Intent use FLAG_IMMUTABLE.

查了一下：
[https://stackoverflow.com/questions/77275691/targeting-u-version-34-and-above-disallows-creating-or-retrieving-a-pendingin](url)

错误表明，从 Android 14（API 34）开始，如果您在具有 FLAG_MUTABLE 的 PendingIntent 对象内有一个隐式意图，您的应用就会崩溃。

如果您不需要FLAG_MUTABLE标志，则只需将其更改为FLAG_IMMUTABLE。但是，如果您的应用需要可变意图，那么我建议采用以下解决方案之一：
val implicitIntent = Intent("com.example.app.action.WHATEVER")
val implicitPendingIntent = if (Build.VERSION.SDK_INT >= 34) {
  PendingIntent.getBroadcast(this, 0, implicitIntent, PendingIntent.FLAG_MUTABLE or PendingIntent.FLAG_ALLOW_UNSAFE_IMPLICIT_INTENT)
} else {
  PendingIntent.getBroadcast(this, 0, implicitIntent, PendingIntent.FLAG_MUTABLE)
}

可以更新一下嘛？

mahmoudramzy400, 2024-06-10:
I also have this issue please solve this ASAP.

lolosssss, 2024-06-17:
https://github.com/jiangdongguo/AndroidUSBCamera/blob/ebfd9437eb6e23e30f583a0151474b2fa6267fca/libuvc/src/main/java/com/jiangdg/usb/USBMonitor.java#L181

Use `PendingIntent.FLAG_IMMUTABLE` instead of custom `PENDING_FLAG_IMMUTABLE`

And also change this: 
https://github.com/jiangdongguo/AndroidUSBCamera/blob/ebfd9437eb6e23e30f583a0151474b2fa6267fca/libuvc/src/main/java/com/jiangdg/usb/USBMonitor.java#L189

to:

```Java
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU) {
	context.registerReceiver(mUsbReceiver, filter, Context.RECEIVER_NOT_EXPORTED);
} else {
	context.registerReceiver(mUsbReceiver, filter);
}
```

Hope this help.

Utsav-Mehta, 2024-07-02:
@lolosssss even this is not helping. Any other solutions?

Utsav-Mehta, 2024-07-05:
@darkSuperman are you able to find any solutions?

xiaohua717, 2024-08-22:
> @lolosssss even this is not helping. Any other solutions?
Have you solved it yet？

Thejas775, 2024-09-07:
@lolosssss Any updates on this ?

xiaohua717, 2024-09-11:
``` java
Intent intent = new Intent(ACTION_USB_PERMISSION);
intent.setPackage(context.getPackageName()); // add it
mPermissionIntent = PendingIntent.getBroadcast(context, 0, intent, PendingIntent.FLAG_MUTABLE);
final IntentFilter filter = new IntentFilter(ACTION_USB_PERMISSION);
 filter.addAction(UsbManager.ACTION_USB_DEVICE_DETACHED);
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU) {
    context.registerReceiver(mUsbReceiver, filter, Context.RECEIVER_NOT_EXPORTED);
} else {
  context.registerReceiver(mUsbReceiver, filter);
 }
```
work for Galaxy S23 , Android 14

Thejas775, 2024-09-11:
> xiaohua717

> ```java
> Intent intent = new Intent(ACTION_USB_PERMISSION);
> intent.setPackage(context.getPackageName()); // add it
> mPermissionIntent = PendingIntent.getBroadcast(context, 0, intent, [...]

> ```java
> Intent intent = new Intent(ACTION_USB_PERMISSION);
> intent.setPackage(context.getPackageName()); // add it
> mPermissionIntent = PendingIntent.getBroadcast(context, 0, intent, [...]

Hi I edited the file but I'm unable to compile. Im a Newbie to java and other stuff. So can you please help me. Thanks

Thejas775, 2024-09-17:
> https://github.com/jiangdongguo/AndroidUSBCamera/blob/ebfd9437eb6e23e30f583a0151474b2fa6267fca/libuvc/src/main/java/com/jiangdg/usb/USBMonitor.java#L181
>
> Use `PendingIntent.FLAG_IMMUTABLE` instead [...]

@jiangdongguo please check on this.
\

SantoshInchalakaranji, 2025-04-05:
@Thejas775 I have compiled but its getting 
`Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x379f7b064183b8`

how to fix this?

SantoshInchalakaranji, 2025-04-05:
@xiaohua717 Can you please share the .aar of libausbc module?

Thejas775, 2025-04-09:
> [@Thejas775](https://github.com/Thejas775) I have compiled but its getting `Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x379f7b064183b8`
> 
> how to fix this?

I used another library

SantoshInchalakaranji, 2025-04-14:
@Thejas775 Can you please share the other library. Thanks

Thejas775, 2025-04-15:
> [@Thejas775](https://github.com/Thejas775) Can you please share the other library. Thanks

https://github.com/quantum6/Android-USB-OTG-Camera/tree/master

This one