jiangdongguo/AndroidUSBCamera #670: 获取不到nv21格式的数据

Nuyoahhjh, 2024-06-25:
这是一个flutter插件，我通过下面的方式获取原始数据，但只返回了rgba数据，请问如何返回nv21格式的数据？
`private fun setPreviewDataCallBack() {
        val callback =  object : IPreviewDataCallBack {
            override fun onPreviewData(data: ByteArray?, width: Int, height: Int, format: IPreviewDataCallBack.DataFormat) {
                if (data != null) {
                    val args = hashMapOf<String, Any>(
                        "data" to data,
                        "width" to width,
                        "height" to height,
                        "format" to format.name
                    )
                    Handler(Looper.getMainLooper()).post {
                        mChannel.invokeMethod("onPreviewData", args)
                    }
                }
            }
        }
        getCurrentCamera()?.addPreviewDataCallBack(callback)
        previewDataCallback = callback
    }`

Qdafengzi, 2024-10-13:
@Nuyoahhjh 
我为何获取不到RGBA的数据 ,都是NV21的 如何获取的rgba?

juechenshiai, 2024-12-17:
setRenderMode(CameraRequest.RenderMode.NORMAL)的时候返回的是uvc的回调数据，格式为nv21，setRenderMode(CameraRequest.RenderMode.OPENGL)的时候返回的是OpenGL的渲染数据，格式为rgba