jiangdongguo/AndroidUSBCamera #677: 使用带mic的camera如何实现显示camera preview画面的同时播放UAC音频？

zhuangwf2017, 2024-07-26:
使用带有mic的camera，3.2.0版本可以同时观看preview画面和播放UAC音频，但是最新版本3.3.3反而不行，虽然录出来的视频和音频文件是有声音的，但不能播放音频，点那个小喇叭图标没反应，请问3.3.3上面怎么改能实现播放音频？

zhuangwf2017, 2024-08-18:
在最新 3.3.3 版本上，点小喇叭图标时，从 logcat 看到，在 "libausbc\src\main\java\com\jiangdg\ausbc\encode\AACEncodeProcessor.kt" 的 initAudioTrack 里面，获取到的 channelConfig 不正确，导致音频播放失败：

07-24 08:33:23.826  6382  6779 I AUSBC   : [AudioSystem]  startRecording success-->3
07-24 08:33:23.827  6382  6778 I AUSBC   : [AACEncodeProcessor]  initAudioTrack: sample=8000,format=2,count=1
...
07-24 08:33:23.827  6382  6778 E android.media.AudioTrack: Front channels must be present in multichannel configurations
07-24 08:33:23.827  6382  6778 E android.media.AudioTrack: getMinBufferSize(): Invalid channel configuration.
07-24 08:33:23.829  6382  6778 E android.media.AudioTrack: Front channels must be present in multichannel configurations
...
07-24 08:33:23.830  6382  6778 E AUSBC   : [AACEncodeProcessor]  start/stop play mic failed, err = Unsupported channel mask configuration 16 for encoding 2
07-24 08:33:23.830  6382  6778 E AUSBC   : java.lang.IllegalArgumentException: Unsupported channel mask configuration 16 for encoding 2
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at android.media.AudioTrack.audioParamCheck(AudioTrack.java:1726)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at android.media.AudioTrack.<init>(AudioTrack.java:801)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at android.media.AudioTrack.<init>(AudioTrack.java:747)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at android.media.AudioTrack.<init>(AudioTrack.java:701)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at android.media.AudioTrack.<init>(AudioTrack.java:642)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at com.jiangdg.ausbc.encode.AACEncodeProcessor.initAudioTrack(AACEncodeProcessor.kt:339)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at com.jiangdg.ausbc.encode.AACEncodeProcessor.playAudioStart$lambda-6(AACEncodeProcessor.kt:156)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at com.jiangdg.ausbc.encode.AACEncodeProcessor.lambda$K0Aqjmwt71ROgJwG6e0PYF8Kj6k(Unknown Source:0)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at com.jiangdg.ausbc.encode.-$$Lambda$AACEncodeProcessor$K0Aqjmwt71ROgJwG6e0PYF8Kj6k.run(Unknown Source:4)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
07-24 08:33:23.830  6382  6778 E AUSBC   : 	at java.lang.Thread.run(Thread.java:920)

试着将
val channelConfig = mAudioRecord.getChannelConfig()
改成
val channelConfig = AudioFormat.CHANNEL_OUT_MONO
也就是将 channelConfig 写死成 AudioFormat.CHANNEL_OUT_MONO，结果点小喇叭图标可以播放音频了，只是音质比较差，毕竟采样率只有 8K，声道数也只有 1。

为什么 mAudioRecord.getChannelConfig() 返回的值不正确？还需要查，目前这么改只是个临时的 workaround。

此时音频播放实际上是通过调用系统录音机 AudioRecord 得到的音频数据（Android 上插入 UAC mic 后系统录音机录音时会缺省从 UAC mic 录制），并不是直接从 UAC 拿数据，
要想直接从 UAC 播放音频，需要修改 "libausbc\src\main\java\com\jiangdg\ausbc\base\CameraFragment.kt"，
将
.setAudioSource(CameraRequest.AudioSource.SOURCE_SYS_MIC)
改成
.setAudioSource(CameraRequest.AudioSource.SOURCE_DEV_MIC)

此外，还需要修改 "libausbc\src\main\java\com\jiangdg\ausbc\encode\audio\AudioStrategyUAC.kt"，
将函数 getChannelConfig 中的 AudioFormat.CHANNEL_IN_MONO 和 AudioFormat.CHANNEL_IN_STEREO 分别改为 AudioFormat.CHANNEL_OUT_MONO 和 AudioFormat.CHANNEL_OUT_STEREO，
原代码这里应该是写错了。

但是点小喇叭图标仍然没反应，分析 logcat 可以看到还是在 initAudioTrack 里面，获取到的采样率 sampleRate 是错误值 -1，导致音频播放失败：

07-28 17:19:12.009 12421 12514 D USBAudio: before strdup vid:11145 pid:22629
07-28 17:19:12.009 12421 12514 D USBAudio: before libusb_init2 ret:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
07-28 17:19:12.009 12421 12514 I libusb/core: [12514*core.c:2085:libusb_init2]:libusb v1.0.19.10903
07-28 17:19:12.010 12421 12514 I libusb/usbfs: [12514*android_usbfs.c:512:op_init2]:call android_scan_devices
07-28 17:19:12.010 12421 12514 D USBAudio: before libusb_get_device_with_fd vid:11145 pid:22629
07-28 17:19:12.011 12421 12514 D USBAudio: open device err LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
07-28 17:19:12.011 12421 12514 D USBAudio: before scan_audio_interface ret:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
07-28 17:19:12.011 12421 12514 D USBAudio: scan_audio_interface
07-28 17:19:12.011 12421 12514 D USBAudio: scan_audio_interface :1
07-28 17:19:12.011 12421 12514 D USBAudio: scan_audio_interface :2
07-28 17:19:12.011 12421 12514 D USBAudio: set _speakerInterface
07-28 17:19:12.011 12421 12514 D USBAudio: Get desc rate 48000
07-28 17:19:12.011 12421 12514 D USBAudio:  _speakerInterface 3 _controlInterface 2 mMaxPacketSize 192 _alternateSetting 1
07-28 17:19:12.011 12421 12514 D USBAudio: kernel detach interface_number:3
07-28 17:19:12.011 12421 12514 D USBAudio: claim_interface r:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
07-28 17:19:12.011 12421 12514 D USBAudio: claim_interface r:0
07-28 17:19:12.011 12421 12514 D USBAudio: Select the  bEndpointAddress:130
07-28 17:19:12.011 12421 12514 D USBAudio: Select the altsetting:1, _speakerInterface:3
07-28 17:19:12.038 12421 12514 D USBAudio: set mic config fail 0 libusb:LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED,errno:No data available
07-28 17:19:12.038 12421 12514 D USBAudio: before sample_rate:48000
07-28 17:19:12.041 12421 12514 D USBAudio: host rate is 48000 ,device rate is 5571354
[... cut ...]
从 logcat 里看到，正确的 sample rate 应该是 48000，
试着将
val sampleRate = mAudioRecord.getSampleRate()
改成
val sampleRate = 48000
结果点小喇叭图标可以正常播放音频了，由于是 48K 采样率，双声道，音质好多了。

为什么 mAudioRecord.getSampleRate() 返回的值不正确？可以追查到一个共享库 libUACAudio.so，但这个库没有提供源码，没办法继续查，目前的改法也只是个临时的 workaround。
注意当 AudioSource 选择 SOURCE_DEV_MIC 时，这里的 AudioRecord 并不是系统录音机，而是作者自己实现的 UAC 录音机，
对应的代码在 "libausbc\src\main\java\com\jiangdg\ausbc\encode\audio\AudioStrategyUAC.kt"。

目前 UAC 音频虽然可以播放了，但仍有不少 bug，希望下个版本能完善。

以上是本人对 UAC 音频播放这部分的分析过程，在 Android 14 上验证，还请原作者指正。

JosephXu2, 2026-03-06:
@zhuangwf2017 你好，请问下这个 libUACAudio.so库你有吗。克隆下载源代码没有这个库，怎么生成的。导致我走SOURCE_DEV_MIC找不到这个库崩溃。可以提供一个吗？没有源码可以编译ndk吧？