jiangdongguo/AndroidUSBCamera #684: JCenter repositories are no longer available in Android gradle 8.0 ++

jonMikelGoya, 2024-09-11:
This library use dependencies like 

com.gyf.immersionbar:immersionbar:3.0.0
com.zlc.glide:webpdecoder:1.6.4.9.0

that are located in JCenter repository, AndroidUSBCamera cannot be used in Android project with gradle 8.0 and above because cannot load this libaries.

pimakshay, 2024-12-26:
Importing this dependency eliminates immersionbar error:
`implementation 'com.geyifeng.immersionbar:immersionbar:3.2.2'`

I have still not resolved the issue with the `webpdecoder` dependency error. Did you figure it out?
   > Could not resolve com.zlc.glide:webpdecoder:1.6.4.9.0.

pimakshay, 2024-12-26:
Update: 
Using GLIDE_VERSION='4.16.0' and the corresponding wepdecoder version worked out for me.
```    implementation "com.github.bumptech.glide:glide:${GLIDE_VERSION}"
    implementation "com.github.bumptech.glide:okhttp3-integration:${GLIDE_VERSION}"
    implementation "com.github.zjupure:webpdecoder:2.6.${GLIDE_VERSION}"
```