jiangdongguo/AndroidUSBCamera #688: Old implementation call

Thejas775, 2024-09-23:
I am using implementation 'com.github.jiangdongguo:AndroidUSBCamera:2.3.4' this implementation call due to  which the old library is being fetched. Can anyone give me update on this?

osokindm, 2024-09-24:
You may try to downgrade your gradle version or update library version

Thejas775, 2024-09-24:
> You may try to downgrade your gradle version or update library version

didnt work

osokindm, 2024-09-25:
another option is to remove this dependency and corresponding code, but you will have to add this lib manually

Thejas775, 2024-09-25:
This library when I searched on maven doesn't exist.

AmirMz1, 2025-03-26:
if you do not use **immersionbar** or **webpdecoder ** in your project, you can exclude them in your build.gradle:

Kotlin Gradle (kts):

```
    implementation("com.github.jiangdongguo:AndroidUSBCamera:3.3.3") {
        exclude (group= "com.gyf.immersionbar", module= "immersionbar") // Exclude old ImmersionBar
        exclude (group= "com.zlc.glide", module= "webpdecoder") // Also exclude webpdecoder if needed
    }
```

Gradle:

```
`dependencies {
    implementation ('com.github.jiangdongguo:AndroidUSBCamera:3.3.3') {
        exclude group: 'com.gyf.immersionbar', module: 'immersionbar' // Exclude old ImmersionBar
        exclude group: 'com.zlc.glide', module: 'webpdecoder' // Also exclude webpdecoder if needed
    }`

```

And if you want to use these libraries, you can install them with the desired version:

```
    implementation 'com.github.gyf-dev:ImmersionBar:3.2.2'
    implementation 'com.github.zlc-platform:webpdecoder:2.1.0' 
```