jiangdongguo/AndroidUSBCamera #689: getAllPreviewSizes is not work

QDQ82, 2024-09-24:
version : implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.2.8'

I want to bring the resolution in advance.
However, it returns "null".

Is there a way?

QDQ82, 2024-09-24:
public void startCapture() {
        try{
            if(mUvcStrategy != null) {

                mUvcStrategy.addPreviewDataCallBack(new IPreviewDataCallBack() {
                    @Override
                    public void onPreviewData(@Nullable byte[] bytes, @NonNull DataFormat dataFormat) {
                        System.out.println(dataFormat.toString());
                    }
                });
                mUvcStrategy.startPreview(getCameraRequest(),surfaceView.getHolder());
            }
        }catch (Exception e){

        }

    }
  
  
    private CameraRequest getCameraRequest() {
        return new CameraRequest.Builder()
                .setPreviewWidth(mainWidth)
                .setPreviewHeight(mainHeight)
                .setFrontCamera(false)
                .setContinuousAFModel(true)
                .setContinuousAutoModel(true)
                .create();
    }
    
    I want to enter the resolution, but I don't know the resolution I support, so it's hard to implement.

It must be operated with startpreview to get the resolution.

Is it possible to change the resolution of the value of the cameraarequest after it has already started?

carlosmg95, 2025-08-28:
Did you solve it? I have the same problem.