jiangdongguo/AndroidUSBCamera #695: Could not find com.gyf.immersionbar:immersionbar:3.0.0.

lsmayday, 2024-10-12:
找不到了！！

akshatsinghania, 2024-10-13:
implementation 'com.geyifeng.immersionbar:immersionbar:3.2.2'

farbodpm, 2025-02-09:
> implementation 'com.geyifeng.immersionbar:immersionbar:3.2.2'

where should we add it ?

AmirMz1, 2025-03-26:
if you do not use **immersionbar** or **webpdecoder ** in your project, you can exclude them in your build.gradle:

Kotlin Gradle (kts):

```
    implementation("com.github.jiangdongguo:AndroidUSBCamera:3.3.3") {
        exclude (group= "com.gyf.immersionbar", module= "immersionbar") // Exclude old ImmersionBar
        exclude (group= "com.zlc.glide", module= "webpdecoder") // Also exclude webpdecoder if needed
    }
```

Gradle:

```
`dependencies {
    implementation ('com.github.jiangdongguo:AndroidUSBCamera:3.3.3') {
        exclude group: 'com.gyf.immersionbar', module: 'immersionbar' // Exclude old ImmersionBar
        exclude group: 'com.zlc.glide', module: 'webpdecoder' // Also exclude webpdecoder if needed
    }`

```

And if you want to use these libraries, you can install them with the desired version:

```
    implementation 'com.github.gyf-dev:ImmersionBar:3.2.2'
    implementation 'com.github.zlc-platform:webpdecoder:2.1.0' 
```