jiangdongguo/AndroidUSBCamera #706: camera thread crash

zzy-33, 2024-12-02:
```
12-02 10:47:20.265 28078 31497 F libc    : Fatal signal 11 (SIGSEGV), code 0 (SI_USER) in tid 31497 (camera-17331076), pid 28078 (t.healthydevice)
12-02 10:47:20.552 31538 31538 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
12-02 10:47:20.553 31538 31538 F DEBUG   : Build fingerprint: 'rockchip/rk3399_mtb927/rk3399_mtb927:11/RQ3A.210705.001/eng.sichen.20241113.072046:userdebug/release-keys'
12-02 10:47:20.553 31538 31538 F DEBUG   : Revision: '0'
12-02 10:47:20.553 31538 31538 F DEBUG   : ABI: 'arm64'
12-02 10:47:20.555 31538 31538 F DEBUG   : Timestamp: 2024-12-02 10:47:20+0800
12-02 10:47:20.555 31538 31538 F DEBUG   : pid: 28078, tid: 31497, name: camera-17331076  >>> com.xhsoft.healthydevice <<<
12-02 10:47:20.555 31538 31538 F DEBUG   : uid: 10179
12-02 10:47:20.555 31538 31538 F DEBUG   : signal 11 (SIGSEGV), code 0 (SI_USER), fault addr --------
12-02 10:47:20.555 31538 31538 F DEBUG   :     x0  0000000000000000  x1  0000000000000081  x2  0000000000000001  x3  0000000000000000
12-02 10:47:20.555 31538 31538 F DEBUG   :     x4  0000000000000000  x5  0000000000000000  x6  0000000000000000  x7  8c5ad6b0802a7f46
12-02 10:47:20.555 31538 31538 F DEBUG   :     x8  0000000000000000  x9  0000000000000c00  x10 0000000000000009  x11 0000000000000000
12-02 10:47:20.555 31538 31538 F DEBUG   :     x12 0000000000000265  x13 ea097a15574e3419  x14 003a05fd35b2c7a6  x15 0000000029aaaaf1
12-02 10:47:20.555 31538 31538 F DEBUG   :     x16 0000007a076d1d10  x17 0000007a076634f0  x18 00000076874b6000  x19 b4000078561a0cb0
12-02 10:47:20.555 31538 31538 F DEBUG   :     x20 b4000078561a03c0  x21 b4000078561a03e8  x22 b4000078061647d0  x23 000000000000000c
12-02 10:47:20.556 31538 31538 F DEBUG   :     x24 0000000000000001  x25 b4000078561a0310  x26 0000000000000009  x27 b4000078561a0d5c
12-02 10:47:20.556 31538 31538 F DEBUG   :     x28 b4000076efacfc40  x29 0000007764b6d9f0
12-02 10:47:20.556 31538 31538 F DEBUG   :     lr  00000076f715f34c  sp  0000007764b6d980  pc  00000076f715f1f0  pst 0000000080000000
12-02 10:47:20.562 31538 31538 F DEBUG   : backtrace:
12-02 10:47:20.562 31538 31538 F DEBUG   :       #00 pc 00000000000131f0  /data/app/~~LjtGB79XaCL4gDVn1ZcDAw==/com.xhsoft.healthydevice-3V5Oy09QI3ETEU89d3etiQ==/base.apk!libuvc.so (offset 0x3bfa000) (BuildId: ecfb28a6291beb57fec7298797d8213c772ae2c1)
12-02 10:47:20.562 31538 31538 F DEBUG   :       #01 pc 000000000000c99c  /data/app/~~LjtGB79XaCL4gDVn1ZcDAw==/com.xhsoft.healthydevice-3V5Oy09QI3ETEU89d3etiQ==/base.apk!libusb100.so (offset 0x949e000) (usbi_handle_transfer_completion+332) (BuildId: 0a3128b26d7b110d699b38
305536ccf99ac8ea1b)
12-02 10:47:20.563 31538 31538 F DEBUG   :       #02 pc 0000000000011efc  /data/app/~~LjtGB79XaCL4gDVn1ZcDAw==/com.xhsoft.healthydevice-3V5Oy09QI3ETEU89d3etiQ==/base.apk!libusb100.so (offset 0x949e000) (BuildId: 0a3128b26d7b110d699b38305536ccf99ac8ea1b)
12-02 10:47:20.563 31538 31538 F DEBUG   :       #03 pc 000000000000d618  /data/app/~~LjtGB79XaCL4gDVn1ZcDAw==/com.xhsoft.healthydevice-3V5Oy09QI3ETEU89d3etiQ==/base.apk!libusb100.so (offset 0x949e000) (BuildId: 0a3128b26d7b110d699b38305536ccf99ac8ea1b)
12-02 10:47:20.563 31538 31538 F DEBUG   :       #04 pc 000000000000d060  /data/app/~~LjtGB79XaCL4gDVn1ZcDAw==/com.xhsoft.healthydevice-3V5Oy09QI3ETEU89d3etiQ==/base.apk!libusb100.so (offset 0x949e000) (libusb_handle_events_timeout_completed+600) (BuildId: 0a3128b26d7b110
d699b38305536ccf99ac8ea1b)
12-02 10:47:20.563 31538 31538 F DEBUG   :       #05 pc 000000000000d774  /data/app/~~LjtGB79XaCL4gDVn1ZcDAw==/com.xhsoft.healthydevice-3V5Oy09QI3ETEU89d3etiQ==/base.apk!libusb100.so (offset 0x949e000) (libusb_handle_events+32) (BuildId: 0a3128b26d7b110d699b38305536ccf99a
c8ea1b)
12-02 10:47:20.563 31538 31538 F DEBUG   :       #06 pc 000000000001155c  /data/app/~~LjtGB79XaCL4gDVn1ZcDAw==/com.xhsoft.healthydevice-3V5Oy09QI3ETEU89d3etiQ==/base.apk!libuvc.so (offset 0x3bfa000) (_uvc_handle_events+132) (BuildId: ecfb28a6291beb57fec7298797d8213c772ae2
c1)
12-02 10:47:20.563 31538 31538 F DEBUG   :       #07 pc 00000000000b0c08  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+64) (BuildId: 0a481e8df134382e9d3effff2fce8b74)
12-02 10:47:20.563 31538 31538 F DEBUG   :       #08 pc 00000000000505d0  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 0a481e8df134382e9d3effff2fce8b74)
--------- beginning of system

```

定位到是
libuvc->init_orginial.c里的

```
/** @internal
 * @brief Event handler thread
 * There's one of these per UVC context.
 * @todo We shouldn't run this if we don't own the USB context
 */
void *_uvc_handle_events(void *arg) {
  uvc_context_t *ctx = (uvc_context_t *) arg;

  while (!ctx->kill_handler_thread)
    libusb_handle_events(ctx->usb_ctx);
  return NULL;
}
```

zzy-33, 2024-12-07:
uvcpreview.cpp文件下的，这边容易阻塞住
```
pthread_join(preview_thread, NULL);
```

j1406493495, 2024-12-11:
32 位 so 同样遇到这个问题，64 位 so 正常，楼主解决了吗？

zzy-33, 2024-12-11:
> 32 位 so 同样遇到这个问题，64 位 so 正常，楼主解决了吗？

并没有，改一下closePreivew的逻辑，稍微好一点，但是还是会出现死锁，或者是指向地址无效的问题

zzy-33, 2024-12-16:
> 32 位 so 同样遇到这个问题，64 位 so 正常，楼主解决了吗？

你是什么设备的android主板

j1406493495, 2024-12-16:
> > 32 位 so 同样遇到这个问题，64 位 so 正常，楼主解决了吗？
> 
> 你是什么设备的android主板

RK3288 Android 7.1
只要是用 32 位 so 的都会有这个问题，64 位的暂时没有发现

zzy-33, 2024-12-16:
> > > 32 位 so 同样遇到这个问题，64 位 so 正常，楼主解决了吗？
> > 
> > 
> > 你是什么设备的android主板
> 
> RK3288 Android 7.1 只要是用 32 位 so 的都会有这个问题，64 位的暂时没有发现

我找到最后，发现了是内存不足道情况了，难崩

j1406493495, 2024-12-16:
> > > > 32 位 so 同样遇到这个问题，64 位 so 正常，楼主解决了吗？
> > > 
> > > 
> > > 你是什么设备的android主板
> > 
> > 
> > RK3288 Android 7.1 只要是用 32 位 so 的都会有这个问题，64 位的暂时没有发现
> 
> 我找到最后，发现了是内存不足道情况了，难崩

你用的什么版本依赖？3.3.3？我用的是 java 版本，2.3.7

zzy-33, 2024-12-16:
> 你用的什么版本依赖？3.3.3？我用的是 java 版本，2.3.7

最新的commit，kotlin版，能➕V不？

CaoXiLin, 2025-01-18:
这个问题解决了吗