jiangdongguo/AndroidUSBCamera #721: Could not find com.gyf.immersionbar:immersionbar:3.0.0.

sanjay-dreamcast, 2025-02-26:
Getting below error any solustion?
`Could not find com.gyf.immersionbar:immersionbar:3.0.0.
`

closed 2025-02-26 by sanjay-dreamcast

nxs4246, 2025-03-24:
facing the same issue rn

AmirMz1, 2025-03-26:
> facing the same issue rn

if you do not use **immersionbar** or **webpdecoder ** in your project, you can exclude them in your build.gradle:

Kotlin Gradle (kts):

```
    implementation("com.github.jiangdongguo:AndroidUSBCamera:3.3.3") {
        exclude (group= "com.gyf.immersionbar", module= "immersionbar") // Exclude old ImmersionBar
        exclude (group= "com.zlc.glide", module= "webpdecoder") // Also exclude webpdecoder if needed
    }
```

Gradle:

```
`dependencies {
    implementation ('com.github.jiangdongguo:AndroidUSBCamera:3.3.3') {
        exclude group: 'com.gyf.immersionbar', module: 'immersionbar' // Exclude old ImmersionBar
        exclude group: 'com.zlc.glide', module: 'webpdecoder' // Also exclude webpdecoder if needed
    }`

```

And if you want to use these libraries, you can install them with the desired version:

```
    implementation 'com.github.gyf-dev:ImmersionBar:3.2.2'
    implementation 'com.github.zlc-platform:webpdecoder:2.1.0' 
```