jiangdongguo/AndroidUSBCamera #722: android 14遇到 android.app.StackTrace: New mutable implicit PendingIntent

hiluo, 2025-03-06:
2025-03-06 17:40:25.376 20723-20723 PendingIntent           com.chevipcheck                      W  android.app.StackTrace: New mutable implicit PendingIntent: pkg=com.xxxxx, action=com.serenegiant.USB_PERMISSION.88345595, featureId=null. This will be blocked once the app targets U+ for security reasons.
                                                                                                    	at android.app.PendingIntent.checkPendingIntent(PendingIntent.java:446)
                                                                                                    	at android.app.PendingIntent.getBroadcastAsUser(PendingIntent.java:734)
                                                                                                    	at android.app.PendingIntent.getBroadcast(PendingIntent.java:721)
                                                                                                    	at com.jiangdg.usb.USBMonitor.register(USBMonitor.java:181)
                                                                                                    	at com.jiangdg.ausbc.MultiCameraClient.register(MultiCameraClient.kt:161)
                                                                                                    	at com.jiangdg.ausbc.base.CameraActivity.registerMultiCamera(CameraActivity.kt:145)
                                                                                                    	at com.jiangdg.ausbc.base.CameraActivity$handleTextureView$1.onSurfaceTextureAvailable(CameraActivity.kt:167)
                                                                                                    	at android.view.TextureView.getTextureLayer(TextureView.java:550)

hiluo, 2025-03-06:
Has anyone here encountered this problem? How to solve it?

hiluo, 2025-03-06:
[image]

hiluo, 2025-03-06:
[image]

hiluo, 2025-03-06:
Looks like we need a new version

thiagoflima, 2025-03-18:
Did you find a workaround for this issue? I've tried to proxy this class without success.

esthernoa, 2025-03-24:
Any solution ??????

thiagoflima, 2025-03-25:
The problem itself is easy to fix. What's difficult is recompiling the project—there are too many outdated methods and functions. I forked the project and I'm trying.