jiangdongguo/AndroidUSBCamera #728: Could not find com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.3.3

AuroraQuestDev, 2025-06-18:
Follow the official instruction, I added `jcenter()` to the `build.gradle` under the `android` folder of my `expo` project.

[image]

Then, added `implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.3.3'` to the `build.gradle` under the `android/app` folder of the same project.

Finally, `npx expo run:android` leads to an error says `Could not find com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.3.3`.

AuroraQuestDev, 2025-06-20:
Seems `v3.2.10` is the latest available tag.

[image]

AnyLifeZLB, 2025-06-21:
貌似不维护了，用这个吧 https://github.com/shiyinghan/UVCAndroid

megahertzon, 2025-06-22:
[image]

megahertzon, 2025-06-22:
you can donwload libausbc-3.3.3.aar on this [link](https://repository.liferay.com/nexus/content/repositories/public/com/github/jiangdongguo/AndroidUSBCamera/libausbc/3.3.3/libausbc-3.3.3.aar)

deniscoman, 2025-06-26:
From the link that @megahertzon posted you can download all the modules that you need. I needed some other modules too.
https://repository.liferay.com/nexus/content/repositories/public/com/github/jiangdongguo/AndroidUSBCamera/

FlinnBella, 2025-07-02:
I was able to bypass this recently by using the version at this maven url - [https://mvnrepository.com/artifact/com.github.chenyeju295.AndroidUSBCamera/libausbc](https://mvnrepository.com/artifact/com.github.chenyeju295.AndroidUSBCamera/libausbc) , with this line:

`implementation("com.github.chenyeju295.AndroidUSBCamera:libausbc:3.3.6")
`  in android/app/build.gradle and the corresponding  `repositories{maven{url maven { url 'https://jitpack.io' }}}` in android/build.gradle.

geecko86, 2025-07-06:
If the library is required by another dependency, adding this on top of what @FlinnBella suggested will fix the build process. 

```
// Substitute the missing AndroidUSBCamera dependencydependencySubstitution {
        substitute module('com.github.jiangdongguo.AndroidUSBCamera:libausbc') using module('com.github.chenyeju295.AndroidUSBCamera:libausbc:3.3.6')
}
```

imwhyang, 2025-07-31:
it works! thanks

> I was able to bypass this recently by using the version at this maven url - https://mvnrepository.com/artifact/com.github.chenyeju295.AndroidUSBCamera/libausbc , with this line: [...]

abeerbendary, 2025-08-03:
it works! thanks

> I was able to bypass this recently by using the version at this maven url - https://mvnrepository.com/artifact/com.github.chenyeju295.AndroidUSBCamera/libausbc , with this line: [...]

abeerbendary, 2025-08-04:
> this not work 
> 
> > you can donwload libausbc-3.3.3.aar on this [link](https://repository.liferay.com/nexus/content/repositories/public/com/github/jiangdongguo/AndroidUSBCamera/libausbc/3.3.3/libausbc-3.3.3.aar)
> > 
> > [image]
> `2025-08-03 19:15:46.555  4557-4557  AndroidRuntime          com.example.myapplication            E  FATAL EXCEPTION: main
>                                                                                                     Process: com.example.myapplication, PID: 4557
>                                                                                                     java.lang.NoClassDefFoundError: Failed resolution of: Lcom/jiangdg/utils/XLogWrapper;
>                                                                                                     	at com.jiangdg.ausbc.utils.Logger.d(Logger.kt:38)
>                                                                                                     	at com.jiangdg.ausbc.utils.ActivityStackUtils.pushActivity(ActivityStackUtils.kt:31)
>                                                                                                     	at com.jiangdg.ausbc.base.BaseActivity.onCreate(BaseActivity.kt:36)
>                                                                                                     	at com.example.myapplication.MainActivity.onCreate(MainActivity.java:29)
>                                                                                                     	at android.app.Activity.performCreate(Activity.java:6734)
>                                                                                                     	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
>                                                                                                     	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2619)
>                                                                                                     	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
>                                                                                                     	at android.app.ActivityThread.-wrap12(ActivityThread.java)
>                                                                                                     	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
>                                                                                                     	at android.os.Handler.dispatchMessage(Handler.java:102)
>                                                                                                     	at android.os.Looper.loop(Looper.java:154)
>                                                                                                     	at android.app.ActivityThread.main(ActivityThread.java:6121)
>                                                                                                     	at java.lang.reflect.Method.invoke(Native Method)
>                                                                                                     	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
>                                                                                                     	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:795)
>                                                                                                     Caused by: java.lang.ClassNotFoundException: Didn't find class "com.jiangdg.utils.XLogWrapper" on path: DexPathList[[zip file "/data/app/com.example.myapplication-1/base.apk"],nativeLibraryDirectories=[/data/app/com.example.myapplication-1/lib/arm64, /system/lib64, /vendor/lib64]]
>                                                                                                     	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
>                                                                                                     	at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
>                                                                                                     	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
>                                                                                                     	at com.jiangdg.ausbc.utils.Logger.d(Logger.kt:38) 
>                                                                                                     	at com.jiangdg.ausbc.utils.ActivityStackUtils.pushActivity(ActivityStackUtils.kt:31) 
>                                                                                                     	at com.jiangdg.ausbc.base.BaseActivity.onCreate(BaseActivity.kt:36) 
>                                                                                                     	at com.example.myapplication.MainActivity.onCreate(MainActivity.java:29) 
>                                                                                                     	at android.app.Activity.performCreate(Activity.java:6734) 
>                                                                                                     	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118) 
>                                                                                                     	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2619) 
[... cut ...]
>                                                                                                     	at java.lang.reflect.Method.invoke(Native Method) 
>                                                                                                     	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905) 
>                                                                                                     	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:795) 
> 
> > `

akshay2796, 2025-08-27:
> It seems the root cause is jar and aar files were not uploaded to the `jitpack.io` repository only pom file.
> 
> I found the way to bypass the issue without using the copy package `com.github.chenyeju295.AndroidUSBCamera` Define `https://repository.liferay.com` (this repo contains all files) repository before `jitpack.io` one in the base project gradle file:
> 
> ```
> buildscript {
>    repositories {
>         mavenCentral()
>         mavenLocal()
>         maven { url 'https://repository.liferay.com/nexus/content/repositories/public/'}         // <------
>         maven { url 'https://www.jitpack.io' }
>         google()
> }
> ```
> 
> To resync the artifact optimally I removed the pom-file from cache gradle folder (`.gradle\caches\modules-2\files-2.1\CLASS_PATH`) and removed the descriptor file in (`.gradle\caches\modules-2\metadata-2.XXX\descriptors\CLASS_PATH`)

It works. Thanks a lot.

plzyxy, 2025-12-12:
https://github.com/plzyxy/AndroidUSBCamera/blob/master/libausbc-3.3.3.aar     我就编译了个 aar     自取吧