jiangdongguo/AndroidUSBCamera #735: Support 16 KB page sizes - Google Play compatibility requirement

AudisoDev, 2025-07-18:
Hello, jiandongguo

Please update the libraries after reviewing the error logs below.
Thank you

[Android Studio Logs]

APK app-debug.apk is not compatible with 16 KB devices. 

Some libraries have LOAD segments not aligned at 16 KB boundaries:

lib/arm64-v8a/libUACAudio.so
lib/arm64-v8a/libUVCCamera.so
lib/arm64-v8a/libjpeg-turbo1500.so
lib/arm64-v8a/libnativelib.so
lib/arm64-v8a/libusb100.so
lib/arm64-v8a/libuvc.so

Starting November 1st, 2025, 
all new apps and updates to existing apps submitted to Google Play and targeting Android 15+ devices must support 16 KB page sizes. 
For more information about compatibility with 16 KB devices, visit developer.android.com/16kb-page-size.

ChloeDimen, 2025-08-14:
同样遇到这个问题APK xxxx.apk is not compatible with 16 KB devices. Some libraries have LOAD segments not aligned at 16 KB boundaries:
lib/arm64-v8a/libUVCCamera.so
lib/arm64-v8a/libjpeg-turbo1500.so
lib/arm64-v8a/libmmkv.so
lib/arm64-v8a/libnativelib.so
lib/arm64-v8a/libusb100.so
lib/arm64-v8a/libuvc.so

ChloeDimen, 2025-09-25:
It seems that the library hasn't been maintained for a long time, so I replaced it. t https://github.com/shiyinghan/UVCAndroid, is it suitable or not

Dimen
***@***.***

jenny850113, 2025-09-26:
@ChloeDimen Thank you for your reply. This problem has been fixed after changing the lib and can be used normally.

closed 2026-03-24 by AudisoDev