jiangdongguo/AndroidUSBCamera #744: Support 16 KB Size

chiragjain1991, 2025-09-24:
Hello, jiandongguo

Please update your libraries after reviewing the error logs below.
Thank you

Google Links : https://developer.android.com/guide/practices/page-sizes#build

[Android Studio Logs]

APK app-debug.apk is not compatible with 16 KB devices.

Some libraries have LOAD segments not aligned at 16 KB boundaries:

Libraries that do not support 16 KB:
base/lib/arm64-v8a/libUACAudio.so
base/lib/arm64-v8a/libUVCCamera.so
base/lib/arm64-v8a/libjpeg-turbo1500.so
base/lib/arm64-v8a/libmmkv.so
base/lib/arm64-v8a/libnativelib.so
base/lib/arm64-v8a/libusb100.so
base/lib/arm64-v8a/libuvc.so
base/lib/x86_64/libUACAudio.so
base/lib/x86_64/libUVCCamera.so
base/lib/x86_64/libglide-webp.so
base/lib/x86_64/libjpeg-turbo1500.so
base/lib/x86_64/libmmkv.so
base/lib/x86_64/libnativelib.so
base/lib/x86_64/libusb100.so
base/lib/x86_64/libuvc.so

Starting November 1st, 2025,
all new apps and updates to existing apps submitted to Google Play and targeting Android 15+ devices must support 16 KB page sizes.
For more information about compatibility with 16 KB devices, visit developer.android.com/16kb-page-size.

megahertzon, 2025-12-10:
I have this problem too.

trymrock, 2026-01-16:
I'm having this problem. Are there any USB camera libraries that support compatibility with Google's requirements?